sinara-hw/Kasli #2: Ethernet stops working when multiple EEM used

gkasprow, 2018-05-26:
@marmeladapk found that when multiple Urukul or other EEM are plugged, Ethernet does not work.
The same happens when Kasli is supplied via extension backplane board.
Did you observe such case?
We will investigate it.

marmeladapk, 2018-05-26:
For example in suservo setup without dios link would not set up. On terminal no errors regarding ethernet are printed. It was a problem when 12V current was higher than ~1,5A.

jordens, 2018-05-27:
I have configured more than a dozen systems and at least 6 of them had 8 EEMs (3 Urukul, 1 Zotino, 1 Sampler, 3 DIO). I haven't seen this. From the description it's unclear what exactly you tried and what your configuration is. You'll have to supply versions, numbers etc.

gkasprow, 2018-05-27:
That's why I'm surprised that this occurred in my lab. Kasli was already tested with various configurations and so far nobody reported such effect.

jordens, 2018-05-28:
And you are sure that the hardware, gatware, firmware, bootloader versions match and are what you expect? What about the rest of the Toolchain?

hartytp, 2019-05-07:
@gkasprow @marmeladapk can you reproduce this, or can we close?

hartytp, 2019-09-19:
Can we close this?

closed 2019-10-01 by hartytp