sinara-hw/Kasli #10: sort out grounding...

hartytp, 2018-07-13:


jordens, 2018-12-18:
@hartytp details and context for this? clock or power or switcher or FPGA or panel?

marmeladapk, 2018-12-18:
I think @hartytp meant what you outlined in #26.

hartytp, 2018-12-18:
@jordens not sure why this doesn't have a description. Let's assume that what I meant was
1. Change default population options to avoid floating the MMCXs
2. Add RC components to allow the FP to be floated, but populate the R with 0R so that by default the FP is tied to circuit ground
3. Use nice FPs with EMC gaskets

Essentially, follow the same grounding strategy we're using in other boards now.

jordens, 2018-12-18:
and ground the SMA input as well.

jordens, 2018-12-18:
Let's implement the MMCX and SMA changes ASAP.

marmeladapk, 2019-01-08:
a87c75c

closed 2019-01-08 by marmeladapk