sinara-hw/Kasli #7: [RFC] Bigger heat sink

hartytp, 2018-07-13:
- [ ] Use a larger heat sink, maybe with a fan like the one on KC705. Use a clip on heat sink instead of a glue on one.

cf https://github.com/sinara-hw/Kasli/issues/5 and https://github.com/sinara-hw/Zotino/issues/4

sbourdeauducq, 2018-07-13:
IMO this is the biggest problem with Kasli right now; the current solution of gluing fans and soldering their wires onto other components of the PCB is not nice.

sbourdeauducq, 2018-07-13:
If 2.0 takes a long time, adding a fan power connector would be a minor change and already a welcome improvement.

marmeladapk, 2019-08-21:
Since I had to move EEM connectors because of SFP connector I will also move the FPGA away from EEM connectors. This will allow me to place heatsink with fan like this: https://cofan-usa.com/pdf/30-2304_REV01.pdf

I couldn't find any heatsinks with fans for 23mm x 23mm package. 

If you have a tried heatsink with fan please let me know.

marmeladapk, 2019-09-10:
Closing in favour of #44

closed 2019-09-10 by marmeladapk