sinara-hw/Kasli #9: [RFC] internal power connector

hartytp, 2018-07-13:
consider adding an internal ATX or other 12V power connector

jordens, 2018-07-13:
The backplane connector or the backplane adapter are already made specifically for that. I'd avoid adding a fourth power input option. It's another part on the bin, another piece that must be documented, and another source of confusion and risk.

gkasprow, 2018-07-13:
We can use RA version of one used on Sayma

hartytp, 2018-07-13:
> The backplane connector or the backplane adapter are already made specifically for that. I'd avoid adding a fourth power input option. It's another part on the bin, another piece that must be documented, and another source of confusion and risk.

True.

I'm not fussed about whether or not we have this or not. I raised it now because I was going over the old issues to break the discussions out into individual issues per item where possible, and this was something we'd discussed before.

> We can use RA version of one used on Sayma

You mean on the BP adapter?

sbourdeauducq, 2018-07-25:
I would like most Kasli systems to use an internal power supply:
* The power brick doesn't fit well in a rack.
* Many systems have free space in the crate that the power module could use.
* The current power supply solution looks a bit consumer/amateur.
* If you build a crate with two Kasli and DRTIO, there is a nice solution to transfer data (use the internal SATA connector) but having two power bricks or a DC connector splitter isn't good.

It seems to me that imposing a backplane adapter on those systems is expensive, bulky and slightly difficult to assemble, compared to adding an ATX connector on the Kasli PCB. Unlike ATX connectors, the DIN 41612 connector isn't meant to be wired directly to a small cable, so a PCB adapter would be necessary to do that cleanly.

@jordens What are the risks with that ATX connector? And, to reduce options, I'm actually happy to do away with the front panel power connector, if we have a nice internal power supply solution.

gkasprow, 2018-07-25:
We discussed it already that PSU + mains filter + IEC socket attached to the chassis rear wall is optimal solution. full ATX connector is LARGE.
It's better to add 4 pin Molex connector ( right angle) that we use on Sayma.  
There is idea to merge Kasli with its backplane as single board - cost difference would be negligible.

gkasprow, 2018-07-25:
If you define exact PN of chassis, we can prepare such rear wall with cutout and remaining stuff.
2 or 3 wall lengths would do the job. Maybe one needs to have some other cutouts for i.e. SMA connectors?

jordens, 2018-07-25:
@sbourdeauducq The risks associated with adding an ATX or molex connector (changes in layout, knock on effects due to other parts being moved, things that can go wrong when using that connector, pinout uncertainty, parts supply, manufacturing the pig-tail) are probably not your biggest problem and not what you want to focus on. Have you looked at the regulatory situation in the different countries? What about certification and safety issues?
If you figure out all the issues then it sounds still much simpler and less risky to either have a 8HP wide module with IEC input and 12V barrel output that you jumper at the front panel into Kasli. That does not break other people's use cases and does not worsen the crate cabling situation by adding another set of cables on top of the ribbon + coax set.
I'm also fine with integrating the BP adapter into Kasli and making Kasli a 200mm card (still fully standard) with a Molex connector instead of the dual barrel connectors in the back. Then there is just two powering options instead of potentially four.
@gkasprow The rear wall in most use cases is currently just a 84HP panel. If you are dead set to put a PSU inside the crate covering different crates and use cases, then I'd just make it a 3U/42HP shallow device to operate next to a 3U/42HP filler panel on the back or a 3U/8HP deep device either for the front or the back. The latter could fit fine opposing a short EEM (*DIO,IDC-BNC,HD68-IDC,Clocker).

sbourdeauducq, 2018-07-25:
> It's better to add 4 pin Molex connector ( right angle) that we use on Sayma.

Yes, I meant that connector (it's also called ATX 4-pin I think?) - not the large motherboard one. There are no pinout uncertainties with that one. Or any other simple connector, really.

> There is idea to merge Kasli with its backplane as single board - cost difference would be negligible.

I'm afraid this might cause chassis compatibility problems.

We should definitely avoid getting locked into one chassis PN; there are often sourcing issues with them so it is better to keep many options.

> The rear wall in most use cases is currently just a 84HP panel.

There are many systems that fit in a 42hp crate.

> If you are dead set to put a PSU inside the crate covering different crates and use cases, then I'd just make it a 3U/42HP shallow device to operate next to a 3U/42HP filler panel on the back or a 3U/8HP deep device either for the front or the back. 

These options sound good, except that it could potentially be problematic to insert a device from the back (the card guides are not symmetrical in some chassis).

jordens, 2018-07-25:
The usual 235mm depth crates (DIN: Fischer, Schroff, Vector) would work fine holding a 220mm (standard eurocard) Kasli. I don't care whether the card extends beyond the 160mm rails or whether we would use the 220mm rails.
And I don't think you need card guides at all if the supply is bolted flat to the rear panel. And yes, they are not symmetric. I just rotate them and since the horizontal rails are strictly symmetric that works fine.  Does that not work for you?

hartytp, 2019-05-07:
What's the conclusion here? Does anyone still want an ATX connector or shall we just scrap this idea (in the long run, we may be able to power via a BP connector)? I'm not fussed either way.

gkasprow, 2019-05-07:
We won't have the backplane connector anymore. So right angle Molex connector or dual footprint (Molex / DC Jack) would be useful.

jordens, 2019-05-08:
If the dual footprint is not a problem, then do it. Tight angle 4 pin Molex being default.

marmeladapk, 2019-05-08:
So right angle dual footprint if possible on the back AND front panel DC jack?

jordens, 2019-05-08:
Barrel only to the front and dual to the back with Molex default.
The Molex Is imo not suitable for external connectivity.

marmeladapk, 2019-08-21:
I placed the same connector as in banker + DC jack connector.
[image]

commit 19ecb762 (2019-08-28): - Switched some components to CERN library (mainly LEDs)
- Removed 1,2V LDO
- Switched to 2x2 SFP cage, added SFP3, sharing lines with SATA port
(SFP selected by default) ( closes #41 )
- Placed rear connectors ( closes #9 )
- Removed unused variant
- Added termination to Helper_DCXO on FPGA
- Moved I2C and UART signals to the same bank ( closes #30 )
- Added LED_USER0, switched LED_USER1-3 to front panel LEDs ( closes #38
)
- Changed EEPROM to one with address pins, added jumpers to all I2C
devices on shared bus
- Added I2C testpoints for DCXOs ( closes #43 )
- Added termination to Helper_DCX [... cut]
closed 2019-08-28 by marmeladapk