sinara-hw/Kasli #11: Extend Kasli to 200mm

hartytp, 2018-07-25:
- Essentially, merge the current "Kasli BP adapter" with Kasli.
- moving discussion from #9 

This is something I'd personally be very keen on for the next batch of Kasli. We will make heavy use of those extra EEM slots (SU-servo eats EEMs), and this gets rid of the annoying BP adapter + associated mechanical issues. Most racks I've looked at work fine with 200mm eurocards.

sbourdeauducq, 2018-07-25:
Oh right, I misunderstood what was meant by "merge with its backplane". Then this sounds fine to me.

marmeladapk, 2019-08-21:
I extended Kasli, but not to 220mm. @gkasprow requested that I conserve some space for a CPCI backplane adapter. For EEM8-11 connectors I used the same layout as on backplane adapter for easier routing.

```
                       8  9
0  1  2  3  4  5  6  7 10 11
```

jordens, 2019-08-22:
Fine.
What about reducing the space between the IDC connectors?

1. IME the IDC connectors can only be detached by pulling on the ribbon.
2. The standard way of crimping ribbon cables is such that the ribbon is pointing to the front panel on one end, i.e. "the wrong way".
3. An IDC strain relief clamp that folds the cable can only be used on 8HP cards.

Therefore adding a IDC ribbon strain relief on the Kasli end is convenient and addresses all three constraints.

gkasprow, 2019-08-22:
True. The 3M strain relief IDC assemblies enable higher density packaging of IDC headers. They can be pulled by the ribbon.

marmeladapk, 2019-09-10:
@jordens Reducing the space between IDC connectors is a minor change that requires a lot of work.

jordens, 2019-09-10:
ACK. This was only meant to point out a possible way to save space if you need it.

commit 8ee3f963 (2019-09-19): - Switched pullup resistors ( closes #46 )
- Switched TCA9517 to PCA9306 for truer 0V low level on I2C bus (both near FPGA and near FTDI)
- Removed EN pullup on IC13, increased pulldown strength on EN IC22
- Added PWM control for fan (+2 pins for fan) ( closes #44 )
- Extended PCB to fit 4 additional EEM connectors ( closes #11 )
- Placed SATA connector above SFP cage ( closes #29 )
- Moved JTAG connector to no longer collide with MMCX connector ( closes #23 )
- Moved SMA connector ( closes #33 )
- Switched level translator for a transistor (watch out for logic inversion)
- Added pullups/downs [... cut]