sinara-hw/Kasli #17: BNX022 soldeirng issue

sbourdeauducq, 2018-08-20:
Sometimes the solder blob on a BNX022 pad touches the component lead but does not wet it. This leads to an unreliable connection and a nasty problem where a board passes tests but no longer powers on after shipping, as the vibrations broke the connection.
Might be a problem with the coating of the pad on some BNX022s as it is unusually difficult to make the solder wet when fixing it manually.

gkasprow, 2018-08-20:
Can you take a photo and post it here please?

sbourdeauducq, 2018-08-26:
It seems the coating is ok, the problem is the large metallic part connected to the ground pins inside the BNX022, which sinks heat (the problem happens on GND but I haven't seen it on +12V).

hartytp, 2019-05-07:
@gkasprow what do we do about this?

gkasprow, 2019-05-07:
There is already the thermal relief used for this part so we cannot do much from PCB design perspective.
One must pay attention during inspection and tests.

hartytp, 2019-05-07:
Ack. I guess these kind of issues are best fed back directly to the manufacturer.

closed 2019-05-07 by hartytp