sinara-hw/Kasli #19: WR PLL/ clock recover

hartytp, 2018-10-22:
Issue to discuss how a WR PLL could be implemented on Kasli (not necessarily suggesting that it *should* at this stage).

- As discussed elsewhere, we wish to use a Digitally Controllable Crystal Oscillator (DCXO) for the oscillator, combined with DDMTD for the phase detector.
- The only good choices of DCXO that we've seen so far are: Si549, LMK61e07, Si570. Si570 is a lower cost option with significantly worse noise. We're having issues getting the LMK61e07 to meet its specified performance, debugging is ongoing. Our current preference is the Si549 as it's very well documented and seems to work as expected without surprises.
- All three DCXO candidates use the same footprint and can be supported by population options. I suggest that we do that and pick the Si549 as out default population
- We should maintain support for the Si5324 at least in the next version, both for risk mitigation and to support easier switching to other clock frequencies until we have really good WR design tools.
- We need a fanout after the DCXO to drive the FPGA + EEM MMCXs. Suggest that we make this fanout a 2-input mux and drive one input from the DCXO and the other from the Si5324. One output routes to the FPGA 'MGTREFCLK_0_216' input
- DCXO needs its own dedicated I2C bus to allow updating at max rate
- DDMTD needs a "helper" oscillator. The noise for this isn't important. Can be either a Si549 or an Si570 depending on which is cheaper (does it cost less to have fewer BOM lines or to use simpler components)? The helper needs its own dedicated I2C bus as well. The helper must drive a cc pin on the fpga
- In "master" mode the SMA supplies the reference clock. This clock must not go through the Si5324 to avoid degrading the noise/stability of the reference
- In principle this can route to either a cc pin or to a MGTREF pin. In our tests on the KC705 we've always used a MGTREF pin (that was the easiest to access), but there is no fundamental reason why it has to. However, we have not directly checked the noise/stability when using a standard CC pin. 
- It would be nice to connect the SMA to a MGTREF pin so that in master mode the transceivers can be clocked directly from the SMA instead of the DCXO/WR-PLL. In principle the transceivers can be clocked from a fabric clock, but that doesn't seem to be a "preferred" option
- Currently `MGTREFCLK1_216` is used by the on-board xtal. As I understand it, the reason for doing this is that the GbE needs a 125MHz clock, but some users may wish to run the WR PLL at a frequency different to 125MHz. @jordens @sbourdeauducq ***am I correct in thinking that it's not an option to route the SMA input to this for the WR reference? If so, we need to use the CC input and measure its performance on a KC705***
- The SMA should still feed one input of the Si5324 (as it currently does). This can either be done by having the SMA feed an FPGA cc pin and then connect an FPGA differential output to the Si5324. The FPGA can loop the clock back. There will be a small amount of degradation of the quality of the signal but, given the known issues with the Si5324, that's probably ok -- the Si5324 will not be used for critical clocks anyway. The other option is to add a fanout before the FPGA. That's a nice option anyway as it provides some protection to the FPGA input against signals from the outside world.
- The `CLKREC` signal is not needed by WR and can be left routed to the Si5324

Comments @jordens @sbourdeauducq @gkasprow @marmeladapk 

One issue that I see is that there are few (no?) spare pins on the FPGA. This will require some thought!

WeiDaZhang, 2018-10-29:
> In principle this can route to either a cc pin or to a MGTREF pin. In our tests on the KC705 we've always used a MGTREF pin (that was the easiest to access), but there is no fundamental reason why it has to. However, we have not directly checked the noise/stability when using a standard CC pin.

> As I understand it, the reason for doing this is that the GbE needs a 125MHz clock, but some users may wish to run the WR PLL at a frequency different to 125MHz.

We are trying to determine if the GTP on Kasli can be driven by a reference input through a pair of MRCC pins, without worsening phase noise and stability of the clock transferred through the serial link.
[image]
A test is done as the diagram shown.
The data on the serial link is configured to be periodic, therefore the phase noise of the data itself can be measured. Though the far-end phase noise is probably whiten by the data, the close-in phase noise should reflect the phase noise of the transmitted clock.
The Tx User Clock is driven by `TXOUTCLKPMA`, the stability of which should indicate the wander/jitter caused by the clock path inside the FPGA.

The test result is interesting, the MRCC pins provide better performance on both phase noise and stability comparing to MGTREF pins, - the default and suggested GTP reference input pins.

[image]
[image]

The stability difference between the two reference paths is about a factor of 2, which is not much. The phase noise difference however is rather big, especially between 10Hz to 100Hz (where the loop filter bandwidth of WR and Si5324 lies), around 10dB.
The result basically suggests, for better-recovered clock at the slave end, the master should use the MRCC pins to drive the GTP and send the data, which is quite opposite to what [Xilinx suggested](https://forums.xilinx.com/t5/7-Series-FPGAs/Using-GTGREFCLK-on-V7-GTP/m-p/685353#M15626).

My hypothesis is that MRCC is less decoupled from the core power supply of the FPGA, while MGTREF is less decoupled from GTP power supply.
During the test, the core of the FPGA is basically idle, but the GTP is fairly busy, which makes MGTREF works not as good as MRCC.

We are doing the test with some work loaded to the FPGA core to see if any difference it makes.

Any other suggestions? @gkasprow @sbourdeauducq  Thanks.

sbourdeauducq, 2018-10-30:
My understanding is that Xilinx decoupled the transceiver clock from the fabric network to make it immune to whatever is happening in the fabric, causing noise that can depend on the bitstream and the data the fabric is processing (from power supplies but also crosstalk).
If we re-characterize this, we have to be very careful and do it over a wide range of fabric activity spread across many resources, including switching activity on different clock networks. We should also make sure that the hardware allows us to fall back on the recommended clocking connection if we experience unforeseen problems. The Xilinx docs are pretty clear that the fabric clock input should not be used.

hartytp, 2018-12-07:
> The Xilinx docs are pretty clear that the fabric clock input should not be used.

@sbourdeauducq yes, I think that's essentially what Weida meant (it *can* be used but shouldn't). And, indeed, subsequent tests show that with any activity on the FPGA fabric at all this does not work.

hartytp, 2018-12-07:
Closing in favour of #25

closed 2018-12-07 by hartytp