sinara-hw/Kasli #20: Kasli v1.1 SDRAM init failed :: Read leveling: Small window

jbqubit, 2018-10-31:
Using 3.0.dev+2729.ga9f0d6a3 on a pair of Kasli v1.1. One board consistently passes SDRAM leveling check and the second fails about 4 of 5 boot attempts. Has this been seen before? 

Good board....
```
Bootloader CRC passed
Gateware ident 3.0.dev+2729.ga9f0d6a3.dirty;brittonlaba
Initializing SDRAM...
Read leveling scan:
Module 1:
00000000000000000011111111000000
Module 0:
00000000000000000011111111000000
Read leveling: 21+-4 21+-4 done
SDRAM initialized
Memory test passed
```

Bad board... s/n TS 277/18 002
```
MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 3.0.dev+2729.ga9f0d6a3.dirty;brittonlaba
Initializing SDRAM...
Read leveling scan:
Module 1:
00000000000000000000011111000000
Module 0:
00000000000000000000111111000000
Read leveling: Small window: 1: 21-25 (5)
SDRAM initialization failed
Halting.
```

jbqubit, 2018-10-31:
The ARTIQ commit is 0b43ec4719dd67aa0d839a9e9c69e26783481bdf.

jbqubit, 2018-10-31:
Also see the same error on another Kasli v1.1. 

When I roll back to a much earlier version of ARTIQ I don't see any symptoms on the problematic boards.

```
[     0.000010s]  INFO(runtime): ARTIQ runtime starting...
[     0.003945s]  INFO(runtime): software ident 4.0.dev+1298.ge4d48a78;opticlock
[     0.011007s]  INFO(runtime): gateware ident 4.0.dev+1298.ge4d48a78;opticlock
[     0.018101s]  INFO(runtime): log level set to INFO by default
[     0.023799s]  INFO(runtime): UART log level set to INFO by default
[     0.029956s]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     1.030653s]  INFO(runtime): continuing boot
[     1.311238s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[     5.589166s]  INFO(board_artiq::si5324):   ...locked
[     5.592905s]  INFO(runtime): using MAC address 00-0a-35-03-1e-78
[     5.598875s]  INFO(runtime): using IP address 128.8.195.91
[     5.606311s]  INFO(runtime::mgmt): management interface active
[     5.620600s]  INFO(runtime::session): accepting network sessions
[     5.635685s]  INFO(runtime::session): running startup kernel
[     5.640243s]  INFO(runtime::session): no startup kernel found
[     5.645978s]  INFO(runtime::session): no connection, starting idle kernel
[     6.105487s]  INFO(runtime::kern_hwreq): resetting RTIO
[    35.722324s]  INFO(runtime::mgmt): new connection from 128.8.195.15:41002
```

sbourdeauducq, 2018-11-01:
duplicate of https://github.com/m-labs/artiq/issues/1149

closed 2018-11-01 by sbourdeauducq