sinara-hw/Kasli #24: Beef up 5V USB to 3.3V LDO (IC16)

jordens, 2018-11-08:
Robustify it against >5.5V USB voltage, surges, and discharging EMI capacitors across power supplies.

gkasprow, 2019-05-11:
The FTDI 1V8 core supply is typ. 70mA. It is derived from 3V3. USB PHY (3V3) consumes max 60mA.
So we have 130mA just for the chip, without the load, output buffers, and translators.
130mA*(5V-3.3V) gives 0.2W of losses. So we don't need the switching converter here and single SOT-23 LDO would do the job nicely.
MIC5219-3.3YM5 works up to 12V, has 500mA of max current and costs pennies. 
Its Thja is 220°C/W so even with 200mA of power will work. I saw SOT-23 LDOs on many boards with this chip.
BYP capacitor is not needed. Remove 10uF load cap because it has too low ESR. Leave only 100nF and 22uF output caps.

marmeladapk, 2019-05-22:
I copied the design from Sayma AMC.

closed 2019-05-22 by marmeladapk