sinara-hw/Kasli #26: ground clock in (SMA) and out (MMCX) [v1.1 BOM change]

jordens, 2018-12-18:
I have a couple times now that that floating shield thing is a bad idea, also on crate-internal MMCX wiring. The symptom is 400/800 kHz switcher modulation sidebands on the urukul clock. Shorting the shield on both ends resolved it each time.
This can a pure BOM change and I'd like to have it implemented on all future Kasli v1.1 builds and Kasli v2.0 as well. 

c.f. #5 and maybe #10

hartytp, 2018-12-18:
Agreed. Floating grounds can be the right thing to do in rare cases, but it's not a sensible default option.

hartytp, 2018-12-18:
(I thought we agreed to do this ages ago)

jordens, 2018-12-18:
We agreed. I'm just pushing to rush this into the current boards without waiting for v1.2/v2.0.

marmeladapk, 2019-01-08:
a87c75c

jordens, 2019-01-08:
Thanks. The v1.1.1 version you assigned is a very good idea.