sinara-hw/Kasli #27: SDRAM module "zero window" breakage

sbourdeauducq, 2019-02-25:
This happened after shipping on a Kasli (mounted in a subrack) that was working before, without touching the firmware:
```
Initializing SDRAM...
Read leveling scan:
Module 1:
00000000000000000000000000000000
Module 0:
00000000000111111111111000000000
Read leveling: Zero window: 1: 0-0 (0)
SDRAM initialization failed
```
Has anyone seen this issue before?

marmeladapk, 2019-02-25:
We may have seen something similar when the FPGA wasn't soldered correctly.

sbourdeauducq, 2019-02-27:
Why did that affect the SDRAM in particular? Which solders were broken?

marmeladapk, 2019-02-27:
Part of the FPGA was lifted and some FPGA pins (particularly lower left quadrant) didn't make proper contact. It was easy to see visually, FPGA wasn't aligned horizontally with the board. We didn't check pin by pin as it was obvious that the FPGA had to be resoldered.

sbourdeauducq, 2019-03-01:
That failing Kasli also developed breakage on EEMs 9 and 10.

jordens, 2019-03-01:
I have one where SFP0 is not receiving and where this SDRAM zero window happens sporadically. It's also a speed grade 3.

hartytp, 2019-05-07:
@gkasprow what do we do about this?

gkasprow, 2019-05-07:
This happened on a PCB series where FPGA was soldered later on manually.
There is nothing to do with PCB design. The tests were probably not sufficient or were not conducted at all.

closed 2019-05-07 by hartytp