sinara-hw/Kasli #29: rotate SATA connector

hartytp, 2019-05-07:


marmeladapk, 2019-05-08:
This also requires rotating JTAG connector -- I'll switch it to surface mounted (same as in Sayma), otherwise it will block current in power planes on all layers.

jordens, 2019-05-08:
What about ditching the JTAG connector? Is it important to keep it? Is it needed and used?

marmeladapk, 2019-05-08:
Yes, for example during tests. Also you cannot use USB JTAG with Vivado (MIG and IBERT) (we'll try something with FTDI in the near future but we're not certain that it will work).

jordens, 2019-05-08:
Ok.
But really? That would imply that you can't use the digilent castellated modules (which are just FTDI) on the standard dev kits (KC705...) with IBERT or MIG.

marmeladapk, 2019-05-08:
Yes you can, however they have special ROM contents and are seen differently by Vivado. And these FTDIs on digilent modules are FT2232 not 4232.

marmeladapk, 2019-08-28:
SATA connector will be rotated and placed above SFP cage.

commit 8ee3f963 (2019-09-19): - Switched pullup resistors ( closes #46 )
- Switched TCA9517 to PCA9306 for truer 0V low level on I2C bus (both near FPGA and near FTDI)
- Removed EN pullup on IC13, increased pulldown strength on EN IC22
- Added PWM control for fan (+2 pins for fan) ( closes #44 )
- Extended PCB to fit 4 additional EEM connectors ( closes #11 )
- Placed SATA connector above SFP cage ( closes #29 )
- Moved JTAG connector to no longer collide with MMCX connector ( closes #23 )
- Moved SMA connector ( closes #33 )
- Switched level translator for a transistor (watch out for logic inversion)
- Added pullups/downs [... cut]
closed 2019-09-19 by marmeladapk