sinara-hw/Kasli #33: move clock SMA

hartytp, 2019-05-08:
> Move the clock input SMA further towards the panel. Leave just enough gap for the insulating washer. Otherwise (with a gap, the two insulating washers, the panel, and the nut, even without the springy/claw washer) there is not enough thread to fully tighten an SMA connector.

I see a similar issue on Urukul as well, where I am not able to fully tighten the SMAs on some cards unless I fiddle with the stackup.

Now that we're not floating grounds by default, should we just DNP these washers?

jordens, 2019-08-22:
I don't think the plastic washers are the cause of the problem. And if the washers are gone, probably the hole would have to be smaller to properly counter the spring washer and nut. The smaller hole would then always prevent using the plastic washers and AC-coupling the shield. If there are no plastic washers, then the foil on the panel or anodizing will be likely be damaged by the standard metal spring/teeth washers. This would need to be considered. Also if we decide against the use of plastic washers, then we should go all the way and fully ground the shield without the use of 0R jumpers.
If this can be solved with getting the positioning and stackup right, then why not do that?

gkasprow, 2019-08-22:
I managed to:
- move the SMA 0.75mm to the panel
- move the board edge to the panel, leave 0.5mm clearance
- increase the SMA hole diameter by 0.5mm
It should solve the issue.

commit 8ee3f963 (2019-09-19): - Switched pullup resistors ( closes #46 )
- Switched TCA9517 to PCA9306 for truer 0V low level on I2C bus (both near FPGA and near FTDI)
- Removed EN pullup on IC13, increased pulldown strength on EN IC22
- Added PWM control for fan (+2 pins for fan) ( closes #44 )
- Extended PCB to fit 4 additional EEM connectors ( closes #11 )
- Placed SATA connector above SFP cage ( closes #29 )
- Moved JTAG connector to no longer collide with MMCX connector ( closes #23 )
- Moved SMA connector ( closes #33 )
- Switched level translator for a transistor (watch out for logic inversion)
- Added pullups/downs [... cut]
closed 2019-09-19 by marmeladapk