sinara-hw/Kasli #35: add note about grounding

hartytp, 2019-05-08:
Add note on schematic by the connectors saying something like "To float connector grounds, replace 0R with 100k. NB Kasli must be connected to mains ground for to avoid damage to it or other equipment connected to it!"

jordens, 2019-05-08:
Please leave out that scare comment about the ground connection. As pointed out elsewhere it's inaccurate, confusing, and inconsistent. It doesn't need to be connected to "mains ground". Your instrument/measurement ground/trap ground will work. It will also work fine if not connected to mains ground.
For that comment to make sense it would need to be clear why it needs to be connected to mains ground, what exactly needs to be connected and why/how not connecting it to mains ground will damage it or other equipment.
It also contradicts itself by implying that the option explicitly described in the sentence before must not be used.

dhslichter, 2019-05-08:
How about "Kasli power input must be connected to a reference ground to prevent switching power supplies from floating to hazardous voltages." or something like that.