sinara-hw/Kasli #38: move some signals to I2C extenders

hartytp, 2019-05-08:
@marmeladapk 
> Here's the list of signals I'd like to move to I2C extenders (and route interrupt signals to the FPGA instead):
> - SFP 0,1,2 (TX_DISABLE, RATE_SELECT, RATE_SELECT1, TX_FAULT, MOD_PRESENT, LED, LOS)
> -led_user1-3
> - VUSB_PRESENT

@jordens 
> Ok moving the signals to an extender apart from the three SFP LEDs please. The other user_leds can go away.

marmeladapk, 2019-06-12:
Additional signals: 
- Clock source select for AD948
- Main, Helper DCXO OE

There are 7 free pins. @jordens requested some additional FP leds, how many do you want?

jordens, 2019-06-12:
I'd like at least one FP LED on a non-I2C pin directly on the FPGA. That LED doesn't need to be associated with a connector. This is really useful for debugging and status signalling on Kasli.
The other LEDs don't matter. One I2C LED per SFP port is fine IMO. Maybe another one associated to the SMA connector.

marmeladapk, 2019-06-12:
@jordens You have 3 SFP LEDs on FP, is that enough or do you want additional one?

jordens, 2019-06-12:
If they go directly to fpga pins then that's fine. if they go through i2c extenders, I'd like another fp led that goes directly to the fpga.

marmeladapk, 2019-06-12:
You asked for SFP LEDs to go directly to the FPGA.

jordens, 2019-06-12:
Good. Then I see no urgent need for more LEDs.

Are the 7 extra pins on the FPGA or on the I2C extender?

marmeladapk, 2019-06-12:
11 on FPGA, 7 on extender, however I would avoid adding more signals to the FPGA just for the sake of it.

marmeladapk, 2019-06-12:
@jordens from https://github.com/sinara-hw/Kasli/issues/39
> And with the I2C extenders we could add a few (maybe two) more edge LEDs to indicate not SFP-related status but system status like e.g. clock lock, idle kernel running, something else running.

Is this still valid then?

jordens, 2019-06-12:
Sorting my thoughts: If the extender pins are free and if there is space to place FP LEDs so that it is clear that they are not related to the SFPs, I think we should clean this situation up a bit and put the LEDs that are associated with the SFPs onto the extender as well and add (two) general purpose indicator LEDs on the FPGA pins to the panel.

marmeladapk, 2019-08-27:
@jordens after adding SFP3 I have one free pin on GPIO expander. What about moving 2 of 3 user leds to FP and adding another one?

jordens, 2019-08-27:
Good

commit 19ecb762 (2019-08-28): - Switched some components to CERN library (mainly LEDs)
- Removed 1,2V LDO
- Switched to 2x2 SFP cage, added SFP3, sharing lines with SATA port
(SFP selected by default) ( closes #41 )
- Placed rear connectors ( closes #9 )
- Removed unused variant
- Added termination to Helper_DCXO on FPGA
- Moved I2C and UART signals to the same bank ( closes #30 )
- Added LED_USER0, switched LED_USER1-3 to front panel LEDs ( closes #38
)
- Changed EEPROM to one with address pins, added jumpers to all I2C
devices on shared bus
- Added I2C testpoints for DCXOs ( closes #43 )
- Added termination to Helper_DCX [... cut]
closed 2019-08-28 by marmeladapk