sinara-hw/Kasli #47: EEM SI simulation

jordens, 2019-09-19:
I'd be interested in seeing the simulated SI of a EEM output from Kasli. Including the FPGA, IDC connectors, 1m ribbon cable, another IDC connector, random patterns on adjacent aggressor channels.

This would be useful for Fastino and for the Urukul SYNC issue.

@hartytp did some nice measurements in #14

gkasprow, 2019-09-19:
I'm afraid it would be hard to find S-models of IDC connectors and assemblies. One can theoretically simulate using connector step model as the geometry but I don't know material characteristics to make it practical.
The easiest way to go would be measurements with a high-speed scope and diff probe.

hartytp, 2019-10-01:
Closing this, as it seems like nothing is going to happen here, but please do reopen if you have further requests along these lines.

closed 2019-10-01 by hartytp