sinara-hw/Kasli #49: LVDS DCXOs

hartytp, 2019-10-01:
@marmeladapk AFICT the LVDS Si549 currently has good availability, while the CMOS ones are hard to find. Shall we switch to the LVDS parts (needs RC changes as well)?

marmeladapk, 2019-10-16:
I made a second variant with LVDS Si549:
[image]
This the only LVDS Si549 that I found to be available. I also DNPed helper termination in this variant near FPGA.

hartytp, 2019-10-16:
@WeiDaZhang can you look at this and sign off asap please? (Check terminations, coupling, voltage levels, part numbers, component availability, etc).

hartytp, 2019-10-16:
@marmeladapk were any changes required for the coupling of the helper PLL to the FPGA?

hartytp, 2019-10-16:
other than that, LGTM

marmeladapk, 2019-10-16:
@hartytp I don't think so, at FPGA we'll use LVDS receiver with termination and Si549 has standard common mode output voltage level.

hartytp, 2019-10-16:
cool

WeiDaZhang, 2019-10-16:
@hartytp 
They all look right.

hartytp, 2019-10-16:
Cool.

hartytp, 2019-10-17:
@marmeladapk for LVDS DCXO, don't you need to DNP R225 (the termination on the helper DCXO at the FPGA)?

marmeladapk, 2019-10-17:
@hartytp It is DNPed.

closed 2019-10-17 by marmeladapk

hartytp, 2019-10-17:
Aah, okay, great!

commit 753bd5e6 (2019-10-16): After schematic review
- Added FBs and decoupling to Si549 ( closes #48 )
- Implemented optional galvanic isolation ( closes #50 )
- Removed FPGA mode and DXP DXN jumpers
- Added diode in parallel with J21
- Removed 0.5pF caps completely
- Added inductors and tantalum capacitors on LT3045 inputs
- Added second variant with LVDS Si549 + changed resistors and capacitors mounting accordingly ( closes #49 )