sinara-hw/Kasli #51: osc2

hartytp, 2019-10-02:
Can someone remind me what the purpose of this is?

What would people think about putting an clock fanout buffer to allow the input SMA clock to be routed to the MGT ref instead of the oscillator?

hartytp, 2019-10-02:
> What would people think about putting an clock fanout buffer to allow the input SMA clock to be routed to the MGT ref instead of the oscillator?

Unless the routing is hard, we could also route it to the SI5324.

This isn't strictly necessary, as we've seen that clocks can be routed via the FPGA without significant degradation of the close-in noise (broadband junk will be killed by the Si5324), but just gives a little extra flexibility. If it's a pain then don't bother.

marmeladapk, 2019-10-15:
@hartytp Isn't it needed for startup, before Si5324 is configured? Also during tests you don't have the dependency on Si5324 and fanout. And routing this is more complicated. Let's avoid complicating this design again.

hartytp, 2019-10-15:
Ok I agree that if it’s currently used we should keep it. I thought @sbourdeauducq said it was no longer used, but maybe I’m wrong?

sbourdeauducq, 2019-10-15:
This one?
[image]

It's used for Ethernet, and also for clocking the comms CPU and its peripherals. Ethernet requires a 125MHz clock and the Si5324 does not necessarily output that frequency, and also the Si5324 needs to be initialized before it outputs any signal.

closed 2019-10-15 by marmeladapk