sinara-hw/Kasli #53: schematic review

hartytp, 2019-10-02:
@marmeladapk I looked at the schematics. They look good!

I've raised a few issues, mainly with questions. Other that that LGTM. NB however I haven't done the kind of thorough review I've done for the EEMs (checking every pin on every IC, reference designs, etc), I assume that wasn't expected for this release.

@sbourdeauducq @WeiDaZhang can you sign off on the WR parts?

@jordens @sbourdeauducq are you happy to sign off on the design over all?

gkasprow, 2019-10-02:
- [ ] is there any reason to keep the SATA connector? 
- [ ] what about removing boot mode selection resistors? Are there any plans to change it?
- [ ] the same with DXN, DXP resistors
- [ ] I'd replace L23, L35 with 1..2.2u inductors. FB has near 0R impedance at the buck frequency. Don't forget about adding poor tantalium to dump LC oscillations.
- [ ] I2C_3V3_SDA/SCL should be called I2C_3V3_SW_SDA/SCL
- [ ] add ordinary diode in parallel to J21. This is to protect the MOSFET and fan
- [ ] 0.5pF caps are not needed with XFMR_MINI-CIRCUITS_TC2-1TX+
- [ ] connect pin 6 of TC2-1TX+ to SMA GND via DNP 0R. At the moment we cannot use isolation potential of the transformer.

sbourdeauducq, 2019-10-02:
>     * is there any reason to keep the SATA connector?

We use it for connecting two Kasli in the same crate with DRTIO. It's a somewhat uncommon use case though.

gkasprow, 2019-10-03:
Now you can do the same with sfp patchcord

hartytp, 2019-10-08:
@jordens @sbourdeauducq have you signed off on this design review?

jordens, 2019-10-16:
Oh. And excellent work on the git commit messages, changelog, and issue tracking @marmeladapk ! Thanks.

hartytp, 2019-10-16:
> Does it make sense to go for the Si53340-B-GM instead of ADCLK948 IC1 and add two more MMCX clock outs? We use it on Urukul already. And it can run at 2.5 V which may make ditching the Si5324 together with its supply more efficient later.

Fine by me. @WeiDaZhang any objections.

@WeiDaZhang please can you confirm that you're happy to sign off on the clocking part of this design asap (apologies if you've already done that!).

Other than that, I'm happy to sign off

hartytp, 2019-10-16:
@WeiDaZhang did you check the rest of the clocking? (I think you did this ages ago, but want to double check you still happy) If so, we just need @sbourdeauducq to sign off on the WR PLL implementation from his side and then I think all reviews are complete.

Would be great to get this off to manufacture in the next week or so

WeiDaZhang, 2019-10-16:
> Fine by me. @WeiDaZhang any objections.

We measured before that ADCLK948 has flicker phase noise of -145dBc/Hz@10kHz offset at 1GHz carrier.
Si53340-B-GM seems to be -137dBc/Hz@10kHz offset at 312MHz, which is slightly worse.

Our WR is no better than 1x10^-13 mod Allan, which about -120dBc/Hz@10kHz offset at 312MHz.
So, should not make any difference.

hartytp, 2019-10-16:
Yes. This is the kind of thing that would have worried me a year ago, but I think we now understand the performance bottlenecks well enough to know that this will not be a limitation (eg compared to the transceiver pll)

@WeiDaZhang are you happy to sign off on the complete design?

WeiDaZhang, 2019-10-16:
> Yes. This is the kind of thing that would have worried me a year ago, but I think we now understand the performance bottlenecks well enough to know that this will not be a limitation (eg compared to the transceiver pll)
> 
> @WeiDaZhang are you happy to sign off on the complete design?

Happy

hartytp, 2019-10-16:
Cool so we just need @sbourdeauducq to sign off on the wr connections and then the review is complete

sbourdeauducq, 2019-10-17:
Except for the support for CMOS Si549 oscillators, this looks OK to me.

hartytp, 2019-10-24:
@marmeladapk Okay, everyone has now completed their reviews. So...when can this go to manufacture?

hartytp, 2019-10-26:
@marmeladapk heads up: the grant we used to pay for our creotech kaslis expires soon so we need to get them to manufacture ASAP to avoid problems

hartytp, 2019-11-23:
Ping @marmeladapk @gkasprow we really need to get kasli 2.0 sent to manufacture in the next week before the grant that paid for the work ends. This is very important to us. In any case, it seems like almost all the work is done already

marmeladapk, 2019-11-28:
@hartytp Layout and schematics are finished.

hartytp, 2019-11-28:
Nice!

I've had a skim over the schematics and it looks good to me. I haven't done the usual thorough design review (check every pin description, eval board/ref design, pin abs max ratings, required capacitances, etc etc) as I'm assuming you did that @marmeladapk 

@WeiDaZhang can you sign off on the WR/clocking please?

@sbourdeauducq @jordens are you happy with this revision? I'd like to send boards to manufacture on monday if possible.

hartytp, 2019-11-29:
@marmeladapk can you remind me which areas had changes? It's worth us carefully reviewing all changes to ensure we don't break anything...

hartytp, 2019-11-29:
I went over the clock page one last time (the only page that has significant changes that I feel qualified to review). Other than cosmetics I don't see any issues.

Any objections to cutting this release on Monday morning?

WeiDaZhang, 2019-11-29:
> I went over the clock page one last time (the only page that has significant changes that I feel qualified to review). Other than cosmetics I don't see any issues.
> 
> Any objections to cutting this release on Monday morning?

May I ask why that?
[image]
Select the power between 2.5/3.3 I suppose.
If so, shall we use P3V3_LDO1 instead then, as Si549s shall not be used at the same time when Si5324 is in use.

hartytp, 2019-12-01:
Thanks @marmeladapk!

I'm assuming everyone has completed their reviews (@WeiDaZhang @sbourdeauducq @jordens ) so we can close this and go to manufacture on Monday once the last straggling issues are closed.

closed 2019-12-01 by hartytp

hartytp, 2020-02-25:
I saw a box labelled Kasli 2 at CTI so I guess this exists now...

hartytp, 2020-02-27:
@marmeladapk we really need these in our lab ASAP. When can you ship them?

marmeladapk, 2020-02-28:
@hartytp I don't even have them on my desk. Do you really want us to send you untested hardware without panels? AFAIK we should receive Kaslis on Monday from production (it's currently during inspection).

Besides you should write to Anna about this not to me, as it's her decision not mine when to ship it.

hartytp, 2020-02-28:
Thanks for the update. Please do test before shipping. Okay, so if receive them on Monday, is it realistic to think that testing will be complete by end of next week?

akaminska, 2020-02-28:
Hi Tom @hartytp , Kasli is still in optical inspection after production. We should get the boards for testing Tuesday morning. We will keep you posted on the progress.

hartytp, 2020-02-28:
Great! Thanks for the update.

marmeladapk, 2020-03-04:
@hartytp @jordens 

[image]

[image]

This SFP cage is massive.

We're now validating if our tests work correctly with the new hardware.

sbourdeauducq, 2020-03-04:
Looks good!
Where is the fan plugged?

marmeladapk, 2020-03-04:
J21, above the FPGA. (2-pin header). Currently the pins are too short, but I plan to change them to header with key, which won't allow reverse connection.

hartytp, 2020-03-04:
Awesome, looks beautiful! Good job @marmeladapk

hartytp, 2020-03-04:
When do the panels arrive?

akaminska, 2020-03-04:
On Monday at the latest.

jordens, 2020-03-04:
Re availability of the SFP cages: I noticed that even Cisco would use two 2x1 cages next to each other when they need 4 SFP and not a 2x2.
And what are we going to do with all the unused space when we go to cPCIs...

gkasprow, 2020-03-04:
CPCIS is not an issue because Kasli would be rotated 180degs and the panel can still be 8HP

jordens, 2020-03-04:
I'm not worried about panel space (and for 4 HP, the cooling stack may or may not be in the way as well).
I was wondering whether there is anything useful we can do with all the unused space of the IDC rats nest which occupies half the entire board area.

marmeladapk, 2020-03-04:
Most of that space will still be required to escape from under the bga package. But perhaps we could make Kasli and other CPCIS boards shorter (160mm)?

gkasprow, 2020-03-04:
The idea behind existing Kasli was to add simple adapter that converts EEMs to CPCI.

jordens, 2020-03-04:
I think it will be much better to draw a line and transition a sufficient set of several modules to cPCIs in one go without adapters.

gkasprow, 2020-03-04:
Yes, but for the moment essential part of the community seems to be happy with existing solution

jordens, 2020-03-04:
The community that M-Labs and QUARTIQ serve would very much appreciate a better solution.

marmeladapk, 2020-03-04:
Kasli v1.1 gateware with uart pins moved to new positions.
```
Initializing SDRAM...
Read leveling scan:
Module 1:
00000000000111111111110000000000
Module 0:
00000000000111111111110000000000
Read leveling: 16+-5 16+-5 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000004s]  INFO(satman): ARTIQ satellite manager starting...
[     0.005612s]  INFO(satman): software ident 5.0.dev+848.g2a909839.dirty;satellite
[     0.013025s]  INFO(satman): gateware ident 5.0.dev+848.g2a909839.dirty;satellite
panic at src/libcore/result.rs:945:5: cannot initialize Si5324: "Si5324 failed to ack write address"
```
I also moved I2C SDA. It seems that I2C works at least initially (no I2C switch error). We'll test this further using our test bench.

akaminska, 2020-03-04:
From the point of view of managing hardware production and stocks, focusing on a specific format of cards for the future instead of having two versions of most cards would make life easier

hartytp, 2020-03-04:
Awesome @marmeladapk 

Re cpics: I can see this being what we want long term but would be nice to have a play with the adapter boards and existing eems before making a decision

gkasprow, 2020-03-04:
this is exactly what we do right now. 3 adapters would be needed to serve all EEM boards.