sinara-hw/Kasli #55: float Si549 OE

sbourdeauducq, 2019-10-17:
* The OE signal can be replaced with register 17 to save a few parts.
* Si549 devices ship with a built-in pull-up or pull-down, depending if OE is active-high or active-low, that activates the output. So it can be left floating.

jordens, 2019-10-17:
The mux has limited isolation. It will be necessary to turn of the unused input. If the Si5324 is used, it seems easier to disable the Si549s via the GPIO extender than via Si549 I2C commands.

sbourdeauducq, 2019-10-21:
> it seems easier to disable the Si549s via the GPIO extender

What is the availability of Si549 variants @gkasprow ? Does it help if the board can be stuffed with either a OE-active-high or a OE-active-low model and be indifferent to it ?

gkasprow, 2019-10-21:
It is poor. At the moment, we have only [this one](https://pl.mouser.com/ProductDetail/Silicon-Labs/549BACB001937ABG?qs=%2Fha2pyFadugqTLoD7lBg52326PpvjdEQ3dg%252B7u%252B5CIg%3D)

marmeladapk, 2019-11-28:
It's already implemented and doesn't affect the cost much.

closed 2019-11-28 by marmeladapk