sinara-hw/Kasli #56: CMOS Si549 support

sbourdeauducq, 2019-10-17:
With the Si549 connected for CMOS:
Why is there a parallel 51 ohm termination for the FPGA? This is not recommended by the Si549 datasheet.
When the Si549 feeds the ADCLK, it also has a 51 ohm series attenuator, and cannot see a 50 ohm impedance. Why does it appear to be not a problem here while the FPGA would need termination?

marmeladapk, 2019-11-26:
> Why is there a parallel 51 ohm termination for the FPGA? This is not recommended by the Si549 datasheet.

It terminates the signal, just differently than series termination. BTW I will change this resistor to ~70 Ohm to match trace impedance.

> When the Si549 feeds the ADCLK, it also has a 51 ohm series attenuator, and cannot see a 50 ohm impedance. Why does it appear to be not a problem here while the FPGA would need termination?

ADCLK has built in 50 Ohm resistors. Come to think of it we may change R37 to 30 Ohm. It will still attenuate signal below maximum value.

marmeladapk, 2019-11-26:
Actually @gkasprow noticed, that series termination is preffered because oscillator may not be able to supply enough current into termination. So I'll change helper termination to 30 ohm series and main also to 30 ohm series.

commit 350ef6c5 (2019-11-28): Release
- DNP J26
- Removed SATA connector
- Added cap to IC5
- IC14, IC15: removed 0R to address pins
- Pin swapping Ext8-11
- Swapped pins on GPIO extenders ( closes #54 )
- Added second reset source for I2C switches from FTDI
- Pulled I2C reset signals to ground
- Added 4 testpoints
- Changed SFP cage, sfp dust covers, FPGA to CERN library
- Changed termination for Si549 to series 30 ohm ( closes #56 )
- Changed termination of CKIN_SMA to 68 ohm to reflect trace impedance
- Xpedition-style copper balancing
closed 2019-11-28 by marmeladapk