sinara-hw/Kasli #63: unused clock complementary outputs

hartytp, 2019-11-29:
On IC1 (clock buffer) some of the complementary outputs aren't connected. IIRC it's recommended practice to always bias both halves of LVPECL even when only using one (otherwise it unbalances the output stage).

marmeladapk, 2019-12-03:
@hartytp In v1.0 we did so to save power. https://github.com/sinara-hw/sinara/issues/204#issue-231061933

hartytp, 2019-12-03:
Ok, makes sense. However, reading app notes recently, it seems that this is not recommended practice. I can't quantify how bad it is though.

Can we add DNPd termination resistors? When the boards arrive I'll try to find time to do some testing to determine if it's worth populating them in future.

marmeladapk, 2019-12-03:
I didn't see any significant difference between v1.0 and v1.1 (though in v1.1 there was adclk944).

It might be tricky to add resistors now, I'll check.

hartytp, 2019-12-03:
Did you look at phase noise/PSRR? Anyway, if this can't easily be done then don't bother. If you can do it easily then it seems worth it.

commit 3c3cea08 (2019-12-15): RC2
Close #65
Added DNPed resistors on unused complementary LVPECL outputs - Close #63
Close #61
Close #59
Switched Si549 to CERNlib
MMCX net lengths were equalized (xsignals were broken). Additionally some xsignals for differential clocks were broken as well. After fixing some net lengths were adjusted within differential pairs.
closed 2019-12-15 by marmeladapk