sinara-hw/Kasli #66: SMPS Emissions

hartytp, 2020-02-27:
@gkasprow when we tested Fastino we saw some relatively large spurs which originated from the main SMPS on Kasli (main frequency is 450kHz, but we also saw (sub)harmonics). @pathfinder49 can you post links the the relevant issues with details please?

The spurs seemed to be heavily dependent on grounding, and could be removed by using a P12V SU with very high AC CMMR/filtering (e.g. an isolated bench supply with an additional CMC + capacitors) to power Kasli. This suggests that the SMPS on Kasli generates relatively large conducted emissions.

Since you have a nice EMI tester in your lab gathering dust, shall we have a look at the emissions from Kasli (doesn't need Fastino to do this)?

If we do find that the emissions are high, we should consider improvements for the next revision. The obvious thing would be to put an LC filter at the SMPS input to prevent noise from coupling onto the 12V network. We should also look again at the layout and see if there are easy ways of optimising it for noise...

pathfinder49, 2020-02-27:
Conducted SMPS noise from Kasli is concisely presented in [this summary](https://github.com/sinara-hw/Fastino/issues/56). A less structured discussion when investigation this is found [here](https://github.com/sinara-hw/Fastino/issues/55).

The SMPS also produces radiated noise, which is [discussed here](https://github.com/sinara-hw/Fastino/issues/59).

pathfinder49, 2020-07-07:
I just came across [these Zotino measurements](https://github.com/sinara-hw/Zotino/issues/21#issuecomment-380472116). This looks extremly similar to what I saw in Fastino. I would wager the same Kasli SMPS noise spurs couple into the Zotino outputs.

dhslichter, 2020-07-13:
@gkasprow is there a plan for making changes to the Kasli SMPS design to reduce this noise, as per some of the recommendations above?

gkasprow, 2020-07-16:
Yes, it is on the todo list. It looks like the noise is injected into the input of the SMPS and then distributed via 12V rail.
The EMI filter present on Kasli does the job by isolating the DC inlet from the SMPS, but not the slave modules.
Once the noise is injected to the 12V rail, it is very difficult to remove it because it is coupled with GND via capacitors, also on the slave modules. The easiest way to mitigate the issue is addition of the PI type LC filter at the SMPS input.

hartytp, 2020-07-16:
> The easiest way to mitigate the issue is addition of the PI type LC filter at the SMPS input.

I agree with that. Wasn't sure if there is any point looking again at the layout as well? (e.g. the kind of ground cuts that you have for the +-12V SMPS module we use on the EEMs).

> The EMI filter present on Kasli does the job by isolating the DC inlet from the SMPS, but not the slave modules

I don't think that's the whole story. In many configurations we were able to suppress this noise by adding an additional CMC on the 12V line between Kasli and the 12V supply. That suggests that the EMI filter doesn't fully prevent the 12V noise form coupling onto the supply. However, I agree that adding an extra LC filter by the SMPS is likely a better solution than altering the EMI filter.

gkasprow, 2020-07-16:
Unfortunately, there is no easy way of checking this with existing HW because there is no simple way of cutting the 12V rail at the SMPS input. P12V0 is directly connected with both EEMs and SMPS.
I’d add 4.7uH inductor + ceramic caps + tantalum caps to damp oscillations.
Of course, the EMI filter has limited attenuation capabilities. Once you inject SMPS noise to the GND, it will stay there forever. So we must do our best to not propagate it to the input.

hartytp, 2020-07-16:
That sounds like a good solution to me!

marmeladapk, 2020-09-02:
Is this still an issue that requires fixing?

From what I've gathered you want to:
- add 4.7uH inductor + ceramic caps + tantalum caps on ADP5052 P12V0 input as close to the chip as possible
- add 2.2 uF (X5R) capacitors on both sides of FL*

I can easily do the latter, but the LC filter will require a sizeable effort due to the fact that P12V0 can come from either side of the board and there's not much space there. This will also require current density simulations on P12V0 with EEMs drawing current.

pathfinder49, 2020-09-02:
My reasons for this fixing would be:
*  Kasli injects SMPS noise into the building mains wiring. We should make sure Kasli is EMI regulation compliant. From my observations, mains EMI should be particularly bad when Kasli is connected to a euro-card that drives a grounded load. 
IIRC, this configuration creates an AC ground loop between Kasli and building ground. This ground loop carries SMPS spur currents. The proposed measures break this AC ground loop. Even if Kasli is regulation compliant, it is worth avoiding the mains EMI. There is EMI sensitive research in the vicinity of Kaslis.
*  The SMPS noise forces (all?) EEM modules to use specific grounding configurations. If these are not followed, the SMPS spurs are transmitted to connected loads. This is demonstrated in [this Fastino spur analysis](https://github.com/sinara-hw/Fastino/issues/56). Similar spurs have also [been shown in Zotino](https://github.com/sinara-hw/Zotino/issues/21#issuecomment-380472116)). 

The measurements presented in https://github.com/sinara-hw/Fastino/issues/56 demonstrate why both fixes are important.

dhslichter, 2020-09-02:
From my view, fixing the SMPS noise on Kasli is a very high priority, and we should be willing to tolerate substantial redesign/simulation pain to get it right.  If it's not right, it can potentially make the hardware useless for high-sensitivity applications.  I think adding inductors (and/or ferrites to provide some more damping?) probably has to be part of the solution.

hartytp, 2020-09-02:
@marmeladapk yes, this is still something we should fix.

> add 4.7uH inductor + ceramic caps + tantalum caps on ADP5052 P12V0 input as close to the chip as possible

Adding an LC filter to the SMPS power input should fix the issue.

We did also discuss changes to the layout as well (e.g. slots in the ground plane), but IIRC @gkasprow suggested that we don't do this.

> the LC filter will require a sizeable effort due to the fact that P12V0 can come from either side of the board and there's not much space there. This will also require current density simulations on P12V0 with EEMs drawing current.

I don't currently why the power entry from two pairs of input filters affect this (not implying it doesn't)?

The best solution might be to just scrap the rear power entry so that we can generally improve the 12V routing/PI.

>add 2.2 uF (X5R) capacitors on both sides of FL*

This isn't a great solution IMHO. Better to stop noise from the SMPS getting onto the main 12V power plane than to just beef up the input CMC. IIRC (but @pathfinder49 do correct me if I'm wrong) adding bigger capacitors doesn't really help much on its own, but it does help if an additional CMC is added to the 12V input.

My gut feeling is that if we add the LC filter then these capacitors won't be necessary. If there is space on the PCB then it might be worth adding DNP pads so we can easily add capacitors if they do turn out to be useful.

hartytp, 2020-09-02:
@dhslichter ferrites won't help with the spurs we reported since all the noise we observed was hundreds of kHz. I don't currently have any data to suggest that the SMPSs are generating noise at ~100MHz where a ferrite would be particularly effective.

It's also worth noting the perhaps obvious point that with any level of power filtering one can still find quite large spurs due to grounding, cabling, etc. I also would recommend keeping low noise cards a few HP away from Kasli to minimize radiative coupling.

dhslichter, 2020-09-02:
> I don't currently have any data to suggest that the SMPSs are generating noise at ~100MHz where a ferrite would be particularly effective.

So basically you are saying that there is limited harmonic content in the noise you are seeing?  Sometimes SMPS put out sharp "spikes" with ~MHz periodicity but frequency content that goes much higher because of their sharpness.

hartytp, 2020-09-02:
>So basically you are saying that there is limited harmonic content in the noise you are seeing? Sometimes SMPS put out sharp "spikes" with ~MHz periodicity but frequency content that goes much higher because of their sharpness.

I don't have evidence of fast spikes. Partially because we're looking at Fastino so (depending on how they couple in) they would be attenuated by the filters, and partially because we didn't stick our analyzer there for the higher frequency measurements. Also, the CMC already on Kasli should do a better job of attenuating the HF stuff, but more characterization would would be needed to make claims about it.

Either way, the LC filter should give us good coverage over a wide frequency range.

marmeladapk, 2020-09-04:
@hartytp Did you check this on Kasli v1.1 or 2.0? 

If on 2.0, did you see 2.5 MHz buck in noise? (IC7 in ClockRecovery.SchDoc).

hartytp, 2020-09-04:
only on v1.1. @pathfinder49 can give you more details than me, but on V1.1 we only saw evidence of that one SMPS...

marmeladapk, 2020-09-04:
@hartytp Please check if these changes are what you and @pathfinder49 had in mind.

hartytp, 2020-09-04:
Thanks!

@pathfinder49 @gkasprow can you review?

closed 2020-09-23 by hartytp