sinara-hw/Kasli #69: SMA isolator

gkasprow, 2020-03-19:
I designed and 3D-printed new spacer that eliminates issue with longer SMA connectors
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

gkasprow, 2020-03-19:
We will do the same for Kasli and other boards.

gkasprow, 2020-03-19:
The print time is 3mins, print time of batch of 20 is 25mins

gkasprow, 2020-03-19:
The stl, PRUSA settings and source file:
[MECH.zip](https://github.com/sinara-hw/Kasli/files/4356828/MECH.zip)
use 0.15mm print setting

hartytp, 2020-03-19:
> The print time is 3mins, print time of batch of 20 is 25mins

How would you convert that into a commercial price (factoring in machine usage etc)?

gkasprow, 2020-03-19:
both CTI and TS have 3D printers and the price is negligible when printed in batches.
Anyway, I volunteer to print any quantity that is needed.

gkasprow, 2020-03-19:
One spool of material worth 50$ is sufficient to print thousands of such isolators.

hartytp, 2020-03-19:
Cool

marmeladapk, 2020-09-23:
Nothing to do here, closing.

closed 2020-09-23 by marmeladapk