sinara-hw/Kasli #71: Don't isolate the rf shielding

jordens, 2020-04-09:
Having continuous electrical contact is the point of it.
[image]

akaminska, 2020-04-10:
What you see is the result of a slight overkill of a different problem. The problem is that the SMA connector pins protrude quite a bit from the PCB and should not, according to good practices, touch the panel. Unfortunately we could not find SMA connectors with shorter pins. One possible solution is to cut the pins close to the PCB manually, but we tried it and this process is quite "barbaric" and introduces a lot of stress on the PCB, introducing significant risk of damaging the module. The solution which was used here is Kapton foil insulation between the panel and the pins. Obviously, one could use it only on the part of the panel next to the pins. In the solution above the panel still has electrical contact with the PCB trough the screws and the brackets holding the panel, but I agee that we should introduce a better solution.

jordens, 2020-04-10:
It's not panel-to-PCB but pane-to-panel contact that's broken. That placement of the Kapton tape makes the RF shielded panel useless. I agree that cutting the SMA pins is a bad idea. I see the following options:

1. place sufficiently thick Kapton tape on the PCB and not on the panel lip as I've seen on other boards
2. place the SMA connectors 0.5 mm back if that works with the minimum thread length
3. longer SMA connectors and moving the SMA connector back
4. abandon the RF shielded panels

gkasprow, 2020-04-21:
5. make small cutouts in the RF shield and then secure one side of  it with Kapton tape.