sinara-hw/Kasli #76: Kasli 2.0 premature fan failure

sbourdeauducq, 2020-05-23:
After a few weeks of being continuously powered on, the fan (model pictured) on my Kasli 2.0 no longer spins smoothly, and makes a lot of noise and vibrations.

[image]

gkasprow, 2020-05-23:
The same happened to the fan on my Xilinx devkit. Isn't it the same company who does them ?

sbourdeauducq, 2020-12-31:
Replacement part seems to be OK, even though it is the same P/N and manufacturer.

closed 2020-12-31 by sbourdeauducq

gkasprow, 2021-01-07:
Maybe it's worth adding the same circuit we use in Thermostat that enables speed control and RPM readout. If we don't need to spin the fan 100% it will significantly prolong its lifetime.