sinara-hw/Kasli #77: Direct IO

gkasprow, 2020-06-24:
The discussion about Kasli IO connector started [here](https://github.com/sinara-hw/DIO_BNC/issues/11)

>    We can add some uFL or MMCX connector + passive protection or buffer to the next Kasli revision. There are still 8 free IO pins. Then a low-cost SMA-UFL pigtail would do the job.

>    If we're going to discuss this then we should discuss on Kasli not here
    I'd still prefer to reserve a couple of IO for future use. If we are going to break them out then a header seems to be a better way than an uFL/MMCX. It's still dirt cheap to use a ribbon cable to route to a tiny PCB that mates with the FP

True, a single 2mm 10pin IDC header would be better. We can always make a simple board with a level translator and SMA and attach it to the front panel. We have a place for it close to the 1V8 converter.
[image]

gkasprow, 2020-09-10:
it seems that after adding the SMPS LC filter we don't have the space for the connector. Moreover, the free IO pins are placed on the other side of FPGA so routing them would be tricky without adding more layers.

hartytp, 2020-09-10:
I'd definitely vote not to have this connector if it means more layers.

Is there another location we can fit it in? Otherwise, better to scrap it IMHO. People can always use the IO on an EEM for all the purposes this header was planning to serve (we now have 12 EEMs on Kasli).

marmeladapk, 2020-09-10:
@hartytp If you need some random IO there are 4 testpoints with 2,54mm pitch which are connected directly to the FPGA. Would that be enough?

gkasprow, 2020-09-10:
@marmeladapk can you add to these test points also 2V5/3V3 test points,  spaced by 2.54mm? Make sure the test point holes are large enough (at least 0.9mm) to fit the standard gold pin header. Place them in such way that reversed assembly won't make any harm to FPGA.
In this way, one can insert a single row IDC plug and connect it to the tiny PCB with translator and connectors, attached the front panel.
3V3 in / 3V3 out DC/DC converters are easily [available](https://pl.mouser.com/Power/DC-DC-Converters/Isolated-DC-DC-Converters/_/N-brwkv?P=1z0wb29Z1yxt7f5), so when needed, isolated IO would be feasible
This would solve this issue.

gkasprow, 2020-09-10:
It would be nice to rotate the pads placement by 90deg so the plug won't collide with IDC tapes;

hartytp, 2020-09-11:
> @hartytp If you need some random IO there are 4 testpoints with 2,54mm pitch which are connected directly to the FPGA. Would that be enough?

I'm not pushing for this (IMHO there is plenty of IO already broken out from Kasli to the EEM connectors so I don't see the need for another random connector). But, yes, that sounds sensible to me.

marmeladapk, 2020-09-11:
@gkasprow I can add P2V5 (these pins are P2V5 so it would give a reference voltage). I can rotate it only when pads are in 2x3 configuration, theres too little space there.

In this configuration if you rotate connector then you short P2V5 with GND is that what you had in mind?

[image]

gkasprow, 2020-09-16:
@marmeladapk yes, reversing supply won't harm the FPGA. Add 3V3 pad as well.
Or, maybe it's better to add a keyed 1.27 connector like this. We can keep it as DNP.
[image]

it won't interfere with ribbon tapes

marmeladapk, 2020-09-22:
I don't think that adding another type of connector is a correct choice. I'll leave it as it is. It seems at least one user is satisfied with this solution and the rest had ~2 weeks to chime in.

closed 2020-09-22 by marmeladapk

gkasprow, 2020-09-22:
@marmeladapk did you add 3V3 pad?

marmeladapk, 2020-09-22:
@gkasprow No.

gkasprow, 2020-09-22:
please add so one can supply the driver capable of driving TTL logic.