sinara-hw/Kasli #78: I2C SDA 0.8V when pulled low by Si5324

hartytp, 2020-06-30:
https://chat.m-labs.hk/m-labs/pl/9anbfbwbw3yr8j7xctdsx6isxr

hartytp, 2020-06-30:
@gkasprow @marmeladapk any ideas?

hartytp, 2020-06-30:
(This currently seems to stop some boards booting with ARTIQ).

hartytp, 2020-06-30:
Si5324 specifies a max of 0.4V at 3V3 (3mA)

[image]

hartytp, 2020-07-01:
@gkasprow any thoughts about this? I can send you a board showing these symptoms if that will help you debug?

marmeladapk, 2020-07-02:
@hartytp I'll take a look at it next week.

hartytp, 2020-07-02:
Thanks @marmeladapk !

I have one more Kasli v2.0 at the moment. It also displays this issue. Do you need me to post it to you, or do you have other v2.0 with this problem?

It's possible that there has been some hardware damage to the two boards I have, but it seems unlikely since they both displayed this straight out of the box and I've never seen it before...

marmeladapk, 2020-07-02:
I'll first check our Kaslis, I suspect that they may have the similar 
problem, just not as bad.
W dniu czw, 2 lip 2020 o 11∶51 użytkownik hartytp

dnadlinger, 2020-07-02:
> I have one more Kasli v2.0 at the moment. 

sb0 now has two of our boards, so I'm not sure how many we have left. (Thinking about it, perhaps you meant to ship only one – apologies.)

gkasprow, 2020-07-02:
@hartytp It looks like the pullup current is 6mA or more.
If you enable all I2C ports, the pullup currents add up. 
With 16x10k we get 3.3/625R=5.3mA
We also have 2k2 which adds another 1.5mA.
So, it looks like all I2C ports are enabled which do not make much sense because I2C addresses overlap.

DonaldKellett, 2020-07-03:
Following https://github.com/m-labs/artiq/pull/1480 , the voltage at the SDA pin in the shared I<sup>2</sup>C bus remains at 0.85V, which causes an "Si5324 failed to ack register" error following a command to write to address `0x68`. Please find attached a sigrok trace demonstrating this issue.
[trace.txt](https://github.com/sinara-hw/Kasli/files/4868725/trace.txt)

gkasprow, 2020-07-03:
@DonaldKellett are you sure that multiple ports are NOT enabled on the I2C switches?

hartytp, 2020-07-03:
@gkasprow how would that cause this? The only pull-up connected to shared SDA is R189 isn't it?

hartytp, 2020-07-03:
Never mind, turns out I didn't remotely understand how the I2C switches work...

[image]

I take your point: if you enable lots of channels at once then you get a much stronger pull-up, which would completely explain why this happens (and why it didn't happen with the CTI tests)...

gkasprow, 2020-07-03:
@hartytp there are two types of I2C switches. The ones that let you choose 1 of n outputs (I2C multiplexer). You simply enter the output number to the register and that's done. The ones we use are in fact I2C switches, not muxes.

gkasprow, 2020-07-03:
to avoid such issues in the future, we can increase the pullups at the outputs of switches let's say 10-fold.

sbourdeauducq, 2020-07-04:
@gkasprow As you can see from the I2C logic analyzer trace (open it with sigrok), just before the Si5324 failure, the firmware writes the 0x00 0x08 control bytes to the two respective switches (note that the LA registers the 0.8V as 0). This is supposed to select only one channel - correct? In https://github.com/m-labs/artiq/pull/1480 we tried repeating the control byte, but no effect.

hartytp, 2020-07-04:
@gkasprow if your hypothesis is that the switch is selecting multiple channels at once, can we quickly test that by just shorting a few of the other outputs to ground and seeing if that affects the output? If it doesn’t then that can’t be the issue...

gkasprow, 2020-07-05:
the question is what is the state of the registers before the firmware writes the 0x00 and 0x08...

DonaldKellett, 2020-07-06:
I've just looked into the I2C_SW test points on my Kasli 2.0 board and the voltage readings of the pins are as follows:

- SDA: approx. 1.15V
- SCL: approx. 2.45V

Please find attached the sigrok [trace](https://github.com/sinara-hw/Kasli/files/4876486/trace.txt) and [annotations](https://github.com/sinara-hw/Kasli/files/4876488/annotations.txt) for these test points. As far as I can tell, this trace is more or less the same than the previous one, but I managed to get some interesting annotations from PulseView by selecting "24xx EEPROM" as the stack decoder on top of the I2C decoder.

To reproduce the trace, connect D0, D1 and GND of the logic analyzer to SDA, SCL (in I2C_SW) and GND of the Kasli 2.0 board respectively.

gkasprow, 2020-07-06:
@DonaldKellett this still does not respond to the question I asked. Please check if the other ports are not enabled! Just probe the switch outputs while trying to talk to the Silabs chip.

DonaldKellett, 2020-07-07:
I looked into this again, and, from my understanding of the above discussion, I would need to probe the outputs of the SCx/SDx pins on my board to determine whether more than one of them are active at any given time. So, a quick look at the Kasli schematics indicates that these pins are located on IC14, which is a tiny IC at the back of my board and therefore difficult to solder wires directly on it. I then tried looking at the PCB layout on page 20 of the Kasli schematics to look for appropriate soldering points away from the IC but the PDF does not display the PCB layout very well and I couldn't open the PCB_Kasli.PCBDOC file online in the [Altium 365 viewer](https://www.altium.com/viewer/).

@gkasprow do you happen to know what is the best place to solder on another port on this pcb?

marmeladapk, 2020-07-07:
@DonaldKellett You could probe I2C signals on EEM connectors.

marmeladapk, 2020-07-08:
I probed Kasli v2.0 and compared it with v1.1. Measurments:

v2.0, measured on shared bus (so closest to Si5324):
[image]

Low level from Si5324 during reads is around 830 mV (I measured it with scope).

---------

v2.0, measured on 3V3_SW bus (between switches and voltage translator):
[image]

Low level from Si5324 during reads is around 1.1 V.

---------

v2.0, measured on FPGA I2C bus (between FPGA and voltage translator):
[image]

Low level from Si5324 during reads is around 1.1 V.

---------

v1.1, measured on Si5324 bus:
[image]

Low level from Si5324 during reads is around 0 V.

----------

v1.1, measured on 3V3_SW bus (between switches and voltage translator):
[image]

Low level from Si5324 during reads is around 530 mV.

----------

v1.1, measured on FPGA I2C bus (between FPGA and voltage translator):
[image]

Low level from Si5324 during reads is around 0 V.

| Place | Si5324 bus | Between switches and v. translator | FPGA bus |
|---|---|---|---|
| v2.0 | 830 mV | 1.1 V | 1.1 V |
| v1.1 | 0 V | 530 mV | 0 V |

Artix datasheet specifies low level voltage of 2V5 CMOS to be 0.7 V. So it's only a coincidence that it works on my boards.

Four things changed between v1.1 and v2.0:
1. voltage translator was changed from bus repeater with voltage translation (TCA9517) to voltage translator (PCA9306)
2. pullup resistors on FPGA bus and Si5324 bus were changed from 2k2 to 10k
3. Si5324 now shares bus with 2 GPIO extenders and SFP3
4. Si5324 doesn't have TCA9517 bus repeater on its bus now

See #46 for rationale of those changes.

Initial 830 mV level surprised me. I know that switches are set only to enable only this bus. So Si5324 has to drive 10k || 2k2 || 10k || 10k to ground which is around 1k3. However in v1.0 it had to drive 2k2 || 2k2 to ground which is a slightly stronger pullup. Either way 3 mA sink should be enough.

In v1.1 Si5324 was "shielded" from rest of the I2C by a repeater. To check if 4. changes anything I'll measure v1.0 later this week, but AFAIR there were no issues with Si5324 there.

Sharing bus with other devices probably doesn't matter. After measuring v1.0 I'll check if changing resistors between switches and v. translator from 2k2 to 10k will help.

jordens, 2020-07-08:
Argh.

hartytp, 2020-07-08:
@marmeladapk One thing I don't understand from your measurements is that it looks like there is a 270mV voltage drop across the TCA9548ARGER (830mV low-level on `SHARED_SDA` and 1V1 on `I2C_3V3_SW_SDA`). The max reistance is specified as being 30Ohm (see below). Unless I'm missing something, that would suggest there is 9mA flowing through it, which would suggest that the pull-up is really only a couple of hundred ohms, wouldn't it, or am I misunderstanding things?

[image]

jordens, 2020-07-08:
And the other strange thing is that the FPGA drives below 0.1 V on v1.1 but only to about 0.3 V on v2.0.

marmeladapk, 2020-07-08:
v1.0, Si5324 bus:
[image]

----

v1.0, between switches and repeater:
[image]

----

v1.0, FPGA bus:

[image]

So it seems that lack of repeater (4.) directly before Si5324 is not a problem. Perhaps having two PCA9306 translating voltage to the same bus is problematic? (USB)

I'll continue on Friday.

gkasprow, 2020-07-08:
@marmeladapk did you check if the switches have multiple outputs enabled?

hartytp, 2020-07-09:
> @marmeladapk did you check if the switches have multiple outputs enabled?

@gkasprow I don't think that can be the issue here (although, I agree it's worth sanity checking).

Among other things, my calculation above suggests that 9mA must be flowing through the I2C switch (and, that's based on worst-case assumptions about switch resistance). With 10k pull-ups, that would require 27 channels, which is more than we have....

gkasprow, 2020-07-09:
This could be assembly error - somebody assembled resistors that are 10 or 100x lower value.

hartytp, 2020-07-09:
Also worth checking. Otherwise, I agree with @marmeladapk that the PCA9306 is the obvious next thing to investigate.

gkasprow, 2020-07-09:
PCA9306 adds pure  series resistance to the chain. It's FET based level translator, not re-driver.

hartytp, 2020-07-09:
To satisfy my curiosity, do you have a rough equivalent circuit for how it works?

gkasprow, 2020-07-09:
It's just a FET with its gate connected to EN pin and substrate connected to GND. The resistance depends on the supply voltage. So, it does not regenerate the signal. It only passes the logic low level and clips the logic high level to the lower supply potential.

sbourdeauducq, 2020-07-10:
Not sure if this is related to the original problem, but @hartytp's board that @DonaldKellett was using has developed what looks like a hardware failure with I2C_SW SCL and SDA now permanently stuck low.

hartytp, 2020-07-10:
Is that just on the FPGA side (i.e. failure of IC13), or is I2C_3V3_SW stuck low as well?

jordens, 2020-07-10:
Elaborating on https://github.com/sinara-hw/Kasli/issues/78#issuecomment-655572273, the > 300 mV when the FPGA is driving low (see "v2.0, measured on FPGA I2C bus (between FPGA and voltage translator):") mean that it's sinking more than 18 mA (12 mA slow drive from IBIS).

hartytp, 2020-07-10:
It's somewhat hard to see where all that current could be coming from. Broken level translator? FTDI chip?

marmeladapk, 2020-07-10:
Ok, I found the culprit. As I thought, USB I2C bus is the problem. FTDI seems to drive all its pins actively high after reset. This means that SDA SCL **and** enable pin of voltage translator (which has pulldown) are driven high. Since it's driven actively it sources current when other devices try to drive the bus low. It wasn't a problem before because previously we used active repeater which didn't have a direct connection to shared bus.

Shared bus has 0.7 V when USB is plugged in and power is disabled.

> The  4  channels  of  the  FT4232H  reset  to  4  asynchronous  serial  UART  interfaces. 

Solutions for now:
- disconnect USB cable - FTDI won't be powered and won't drive line high (quickest, though least satisfactory)
- program FTDI to keep these pins low
- remove IC22
- possibly programming EEPROM with "Pull down I/O Pins in USB Suspend" could help but I will have to check if it's reliable (it could work only when nothing is connected to serial terminal)

Long-term solutions:
- Insert inverter on enable pin, so that by default IC22 is disabled (preferred by me)
- Change PCA9306 on this bus back to TCA

hartytp, 2020-07-10:
> Insert inverter on enable pin, so that by default IC22 is disabled (preferred by me)

I don't have the layout in front of me, but is that something we can easily hack into the existing boards with a scalpel and some glue?

jordens, 2020-07-10:
We're already programming the FTDI EEPROM. I bet there is an option to solve this properly.
https://github.com/quartiq/kasli-i2c/blob/master/kasli-ft4232h.conf.in
Try adding `suspend_pull_downs=true`.

jordens, 2020-07-10:
> I don't have the layout in front of me, but is that something we can easily hack into the existing boards with a scalpel and some glue?

Why not just look into programming the EEPROM properly?

hartytp, 2020-07-10:
> Why not just look into programming the EEPROM properly?

If that works reliably then it's fine by me.

marmeladapk, 2020-07-10:
@jordens It may not work if you're connected to the terminal (so FTDI is not USB is not in suspend mode).

sbourdeauducq, 2020-07-10:
I still think the enable pin should be inverted in v2.1 to make the hardware friendlier.

jordens, 2020-07-10:
> @jordens It may not work if you're connected to the terminal (so FTDI is not USB is not in suspend mode).

I meant look for the actual option that sets the state of the FTDI interfaces that are not being used. Note that the four ports are four different and independent USB interfaces. Connecting to one doesn't mean much for the others.

hartytp, 2020-07-10:
@jordens I hadn't seen https://github.com/quartiq/kasli-i2c/blob/master/kasli-ft4232h.conf.in before (I've been using ftprog). That's a nice util.

jordens, 2020-07-10:
> If that works reliably then it's fine by me.

I can't see how properly setting up the EEPROM could be nearly as unreliable as "a scalpel and some glue".

jordens, 2020-07-10:
> I still think the enable pin should be inverted in v2.1 to make the hardware friendlier.

Because of the implied incompatibility this is a bad and shortsighted idea.
Why not just configure it properly?

sbourdeauducq, 2020-07-10:
Removing IC22 is easier than programming FTDI chips, especially since FTDI chips are prone to all sort of bugs.

sbourdeauducq, 2020-07-10:
>  Why not just configure it properly?

Why not just make hardware that is well-behaved by default, especially since the contention may cause permanent damage?

sbourdeauducq, 2020-07-10:
Can't the incompatibility issue be resolved with ~3 lines of code that look at the hardware version and invert the pin?

hartytp, 2020-07-10:
> Removing IC22 is easier than programming FTDI chips

Removing I2C would break all the USB I2C tooling which would be a big regression IMHO

hartytp, 2020-07-10:
Another option is to just add a couple of current limiting resistors between the FTDI chip and IC22. That way we're guaranteed that programming errors can't cause hardware damage (which feels like something we should aim for) at close to 0 extra cost/complexity...

marmeladapk, 2020-07-10:
@jordens 

`suspend_pull_downs` doesn't help regardless if console is opened.

> The EEPROM contents have no effect on the selected mode with the exception of selecting the TXDEN for RS485 mode when asynchronous serial interface has been selected in software. If the device is reset, then the 4 channels must be reconfigured into the required mode.

So programming FTDI cannot fix it by default unless there are other options to set in EEPROM (there aren't to my knowledge).

> I don't have the layout in front of me, but is that something we can easily hack into the existing boards with a scalpel and some glue?

@hartytp It's possible but not easy. If you want to add inverter know, easier fix would be to lift 8th pin of IC22 (EN pin) and add inverter (transistor) on top of it + some wires to pulldown resistor nearby.

> I still think the enable pin should be inverted in v2.1 to make the hardware friendlier.

I'm also for it. It will be a proper fix (and currently seems the only possible permanent one).

marmeladapk, 2020-07-10:
> Not sure if this is related to the original problem, but @hartytp's board that @DonaldKellett was using has developed what looks like a hardware failure with I2C_SW SCL and SDA now permanently stuck low.

@sbourdeauducq This looks like a hardware failure. Probably ESD. Let's hope that it's the voltage translator that's fried, not FPGA inputs.

jordens, 2020-07-10:
Yes. There is no EEPROM fix. I went through the entire thing, including the not-yet exposed bits of `libftdi`. That pin is DTR and is an active low output. My proposal ist the following:

* Disconnect BDBUS4, pin 26 from its via
* Connect the via to BDBUS6, pin 28

DCD is an input with a weak enough pull-up. This is forwards and backwards compatible, doesn't need new hardware, and no version detection and special treatment in software.

hartytp, 2020-07-10:
Any objections to pushing out a v2.1 release with minor fixes soon?

jordens, 2020-07-10:
That seems much to early given that a lot of the board hasn't really been tested thoroughly yet. Little data/solutions on the fan, wr clock recovery, 4 sfp, smps noise, panel mechanics, just to name a few.

marmeladapk, 2020-07-14:
I'll check @jordens proposal the next time I'll be in our lab.

@jordens would you have a moment to check if I2C from FTDI works as expected? Or maybe you have already checked it? (even without the fix for current problem)

[6 more replies]

commit 47d09672 (2020-09-04): Kasli v2.0.1
- Close #80
- Fixed #78
- Change LT3045 output capacitors to 0603 22uF 6.3V, fixed #70
- Fixed #68
- Fixed #67
- #66: added LC filter to SMPS input and fan supply (both IC6 and IC7)
- Changed project name
- Updated copyright year and version on schematics, added license on PCB
- Used buses to simplify LVDS harnesses
- General cleanup of schematics
closed 2020-09-04 by marmeladapk