sinara-hw/Kasli #79: v2.1

hartytp, 2020-07-10:
From https://github.com/sinara-hw/Kasli/issues/78#issuecomment-656806685...

> That seems much to early given that a lot of the board hasn't really been tested thoroughly yet. Little data/solutions on the fan, wr clock recovery, 4 sfp, smps noise, panel mechanics, just to name a few.

Fair enough, but I'd like to move to a new revision soon as this is going to be annoying to patch on v2.0 boards.

>  Little data/solutions on the fan

What data are you looking for here? Do we even have enough v2.0 boards produced to get statistics on that?

Am I right in thinking that there are other fan parts that are mechanically compatible, so we can easily change the fan part as just a BOM change? If so, it's not clear to me that this issue should be a release blocker.

> wr clock recovery

@sbourdeauducq what's your eta for having another look at this?

> 4 sfp

@marmeladapk I assume you did test these post-production? @sbourdeauducq did you check that you could get links on all 4 SFPs? What other testing is needed here?

>  smps noise

What testing do you want to see here? We should have a bit more discussion about options on https://github.com/sinara-hw/Kasli/issues/66. AFAICT all we're talking about would be adding an inductor on the SMPS input to limit noise currents on the main PSU input. It's not easy to prototype that on the current hardware because signals are routed on internal layers. My preference would be to build some prototype boards with the inductors fitted, test those then make a decision. I don't see a good way of doing that with existing hw...

> panel mechanics

Anything beyond 
- https://github.com/sinara-hw/Kasli/issues/72
- https://github.com/sinara-hw/Kasli/issues/69
- https://github.com/sinara-hw/Kasli/issues/71
- https://github.com/sinara-hw/Kasli/issues/74

Does this need testing? 

>, just to name a few.

What else?

sbourdeauducq, 2020-07-11:
> @sbourdeauducq did you check that you could get links on all 4 SFPs?

Yes.

sbourdeauducq, 2020-07-11:
> > wr clock recovery
> 
> @sbourdeauducq what's your eta for having another look at this?

Haven't found hardware problems so far. The problem with ADPLL not being written by the gateware is due to the gateware outputting garbage on the I2C lines, as seen with a logic analyzer.

hartytp, 2020-07-11:
> Haven't found hardware problems so far. The problem with ADPLL not being written by the gateware is due to the gateware outputting garbage on the I2C lines, as seen with a logic analyzer.

Aah, cool! Glad to hear you're making progress on this.

hartytp, 2020-07-11:
@marmeladapk what's your schedule like these days? Do you think you could look at some of the open issues and start to think about resolving them?

I'd really like to get a usable board in the 2.x family relatively soon

marmeladapk, 2020-07-14:
> The problem with ADPLL not being written by the gateware is due to the gateware outputting garbage on the I2C lines, as seen with a logic analyzer.

@sbourdeauducq Some time ago I fixed misoc I2C core during unrelated debugging session. I made version both for CSRs and for wishbone. The code is not polished but it worked. I can send you a diff if you want (and still need it).

> what's your schedule like these days? Do you think you could look at some of the open issues and start to think about resolving them?
> I'd really like to get a usable board in the 2.x family relatively soon

My schedule right now is pretty packed so not really. I could do #78 and #67 but I don't think it warrants an entire production run.

sbourdeauducq, 2020-08-19:
This is fixed now (was just a 1-line change), but thanks.

hartytp, 2020-09-22:
@marmeladapk where are we on the various open issues?

hartytp, 2020-09-23:
All remaining issues are BOM/mechanical changes only, so the plan is to freeze the schematic here so we can move to production.

Thanks for all the work on this @marmeladapk @gkasprow

closed 2020-09-23 by hartytp

dtcallcock, 2020-09-23:
Is this the big production run that everyone will buy and use in their lab or is another round of prototype testing needed?

hartytp, 2020-09-23:
With the caveat that the WR oscillators still haven't been fully tested yet, I believe that this release fixes all known issues. So unless/until new issues turn up this is it.

Kasli is now a pretty well-tested design and there haven't been major changes since v1.0 -- just small improvements here and there -- so I wouldn't personally hold back from ordering

dhslichter, 2020-09-28:
unless you want that Zynq Kasli 😁