sinara-hw/Kasli #81: Rear power entry

hartytp, 2020-09-02:
Do we still want to support power entry at both the rear and front of Kasli? It makes the routing quite a bit harder and I'm not aware of anyone actually using it.

What about keeping the molex connector, but moving it to near the front panel. That way one can get power in from the rear if required (just run a cable from the rear to the front of the PCB, it's a little ugly but no worse than the clock routing) but we don't need two sets of line filters and can simplify the layout.

Any objections @marmeladapk @sbourdeauducq @jordens @cjbe ?

jordens, 2020-09-02:
Ack. Straight molex, e.g. where that ftdi gpio header is.

hartytp, 2020-09-02:
Cool. Unless we hear any objections, let's do that.

marmeladapk, 2020-09-02:
It doesn't matter. 12V still needs to go to EEM connectors to the back so having a connector there doesn't change anything. The only thing that would make routing easier w.r.t filtering discussed in #66 is dropping front panel connector, not the back one.

I'd keep it as it is to avoid change for the sake of change. It's already there, if no one uses it I can DNP components.

Can I close this issue?

hartytp, 2020-09-02:
Aah, ok, I misunderstood your comment in the other issue. The FP connector is the only one we really have to keep, so let's keep the design as is.

closed 2020-09-02 by hartytp