sinara-hw/Kasli #82: DC bias of CKIN_SMA on Kasli2

sbourdeauducq, 2020-09-18:
On Kasli 2.0, the CKIN_SMA pins do not have an external DC bias, which is required for the FPGA differential input buffer to work properly.
The IN_TERM=UNTUNED_SPLIT_50 feature of the FPGA cannot be used to create the bias, because it is not supported at VCCO=2.5V (it is also uncalibrated, and might not be good enough for impedance matching).

gkasprow, 2020-09-18:
We can still fix it in current revision. @marmeladapk

jordens, 2020-09-18:
That would be great.
How feasible is a rework on existing boards? Dc couple and bias the center tap?

gkasprow, 2020-09-18:
I've noticed that generally, FPGA work without bias due to embedded protection diodes that do some kind of DC recovery. However, it's enough to add a 2x10k resistor divider to any of the diff inputs

hartytp, 2020-09-18:
> How feasible is a rework on existing boards? Dc couple and bias the center tap?

Or just leave it AC coupled and use a few RCs to add a bias on the FPGA side of the coupling caps. It looks like there should be room to access the capacitor pads on the current layout.

[image]

marmeladapk, 2020-09-22:
[image]

I removed AC couple and added bias to center tap.

jordens, 2020-09-22:
Does the balun bypass work without a bias and ac-coupled on the `_p` signal? Can be scrapped IMO.

marmeladapk, 2020-09-22:
If you remove balun then you can solder 0R between 1st and 2nd pins giving _P signal DC bias.

closed 2020-09-22 by marmeladapk

hartytp, 2020-09-22:
> If you remove balun then you can solder 0R between 1st and 2nd pins giving _P signal DC bias. I'll add a note about that.

Which pads are you using for that?

sbourdeauducq, 2020-09-23:
I'm not sure if a single-ended signal is a good idea since we lose the comparator present in FPGA differential inputs, and a sine wave into a CMOS input buffer may be noisy.

sbourdeauducq, 2020-09-23:
> However, it's enough to add a 2x10k resistor divider to any of the diff inputs

What are easy soldering locations to do this?

marmeladapk, 2020-09-23:
I'd say the best one is under the FPGA. There are pads for termination resistor there:

[image]

[image]

sbourdeauducq, 2020-09-23:
@marmeladapk thank you

SingularitySurfer, 2020-10-01:
Hi, I did some further debugging because it still didn't work after the mod. In addition, the sl5324 needs to be configured for CKIN1 (currently it's still set to CKIN2 in artiq). But even with this change, the si5324 does not lock, even tho we verified there is a clocksignal at CKIN1 coming from the FPGA. Any ideas?

SingularitySurfer, 2020-10-06:
The LVDS seems to be happy without external bias with a 0dBm clock input signal.

sbourdeauducq, 2020-10-07:
Hmm. I would worry about noise or long-term effects... Are negative voltages/currents within FPGA ratings?

SingularitySurfer, 2020-10-07:
Kinda funny: With external bias to 1.25V (which is 50mV off the 1.2V LVDS center) it's fine until ~ -8dBm.
With no external bias until ~ -14dBm, however once it loses signal you have to go quite a bit higher before it works again.
And ofc. thats just a one off try and not according to spec.

hartytp, 2020-10-07:
> With no external bias until ~ -14dBm, however once it loses signal you have to go quite a bit higher before it works again.
And ofc. thats just a one off try and not according to spec.

Sounds like you're baising by rectifying the input clock through the FPGA ESD capacitors, hence the hysteresis

jordens, 2020-10-07:
Long term this is completely harmless (~50 nC per boot) and I don't see how the rectification (assuming you mean ESD *diodes*?) explains the hysteresis. As Greg days, it explains the self-biasing. I'd look at the si5324 and its lock/LOS hysteresis. The V,T stability of the self-biasing might be interesting though, but even that would be common mode.

commit 44c57284 (2020-09-22): Added DC bias to input clock.
Closes #82