sinara-hw/Kasli #84: Kasli v1.1 suddenly died

sotirova, 2020-10-23:
I had a Kasli v1.1 connected to a Phaser board that I was trying to test. After sitting unattended for a day it suddenly stopped working and none of the LEDs (apart from power) on the kasli light up when it is plugged in.

cjbe, 2020-10-23:
@sotirova can you list exactly which of the power LEDs come on? IIRC only 2 of the power LEDs lit up, the rest of the rails were dead, showing that quite a lot of the power network was misbehaving.

sotirova, 2020-10-23:
The power LEDs that light up when it is plugged in are 1V0 and 12V0.

gkasprow, 2020-10-23:
Did you try to disconnect the Phaser?

sotirova, 2020-10-23:
Yes, it is currently disconnected and only 1V0 and 12V0 light up when powered.

hartytp, 2020-10-23:
What are the voltages on the various rails?

marmeladapk, 2020-10-23:
I had a problem like this once. It's probable that something shorted P2V5 rail. Was Kasli cooled when it sat idly?

Could look with a scope on P2V5?

gkasprow, 2020-10-23:
I have a few Kaslis with broken 2V5 rail - shorted FPGA bank
This happens usually during hot-plug or when 3V3 appears on LVDS lines.
Are you sure nobody touched the setup?

sotirova, 2020-10-23:
Here are the voltages:

- P1V0: 1 V
- P1V2: 0.435 V
- P1V5: 1e-3 V
- P1V8: 9e-3 V
- P2V5: 16e-3 V
- P3V3: 0 V
- P12V0: 12.2 V

marmeladapk, 2021-01-18:
Could you check with multimeter in short detection mode if P3V3 and GND are shorted when board is unplugged? P3V3 is next in power sequence after P1V0.

closed 2022-07-25 by marmeladapk