sinara-hw/Kasli #85: Missing panel design outputs for v2.0 +

HarryMakes, 2020-11-06:
I would like to obtain some outputs for the front panel design, e.g. in form of PDFs like [this one for v1.1](https://github.com/sinara-hw/Kasli/releases/download/v1.1.1/PANELS.PDF). It would be nice to include such outputs for version 2.0 or up on the releases.

gkasprow, 2020-11-06:
You can generate them from sources.

HarryMakes, 2020-11-09:
Hi Greg, thanks for the reply. Since I can't use a Windows OS at work, I tried opening the source files using [Altium's online viewer](https://www.altium.com/viewer/). 

I'd guess the file containing the front panel design would be `PCB_Kasli.PCBDOC`, but with the online viewer I can only see the layout designs of the PCB but not the front panel. There's also a 3D view where I can see most parts of the PCB and components, but the front panel doesn't seem to be complete - it appears cut in half to be 4HP instead of 8HP wide. I took a screenshot as attached below:
[image]

May I get some advice on whether or not `PCB_Kasli.PCBDOC` is the right file that contains the front panel design? Thank you.

gkasprow, 2020-11-09:
Kasli was not designed by me. It was a Creotech job.

HarryMakes, 2020-11-10:
Thanks! I'll reach out to him and write some updates here later on.

KreatorM, 2026-11-04:
@gkasprow Waking up an old thread - I have a github/licence question. For our Kaslis 2.0.2 we have designed new panels - slightly narrower that are distributed by Creotech. Since this is obviously a modified open hardware, according to CERN licence,  should we upload the step files to github? Or this is outside the scope of the open hardware licence?