sinara-hw/Kasli #86: Question about Cu thickness of top/bottom layer

HarryMakes, 2020-11-09:
For Kasli 2.0.1, is it correct that each of the top and bottom layers uses 1.5oz copper in production? 

If there's a manufacturing limitation that doesn't allow 1.5oz for the trace spacing and/or width, would 1oz copper be suitable as a replacement? Are there current or temperature constraints imposing on the thickness of the top/bottom layer Cu? Thanks!

gkasprow, 2020-11-09:
Kasli needs impedance controlled traces. To use other thickness you'd have to redesign the PCB

gkasprow, 2020-11-09:
Also, the current density/drop analysis was performed for a specific thickness. If you use a thinner one, the voltage drop will be higher which may cause issues with stability. It's quite advanced PCB, if you change something, you do it at your risk. The PCB was designed to be compatible with certain PCB material (substrates, prepreg) vendors. Generally, in the case of complex PCB, you cannot change the process without redesigning it.

gkasprow, 2020-11-09:
Moreover, this is basic thickness, the manufacturer adds usually plating. AFAIR it was designed for 56um total thickness.

HarryMakes, 2020-11-11:
Greg, thanks for the info. I've made a reminder that impedance control is applied to all kinds of traces for the board.

Without much PCB manufacturing experience, I would like to know whether or not the 56um total Cu thickness is a parameter that certain PCB standards can satisfy. If I produce the PCB by following IPC class 2 requirements, would the stackup shown in the current Gerber files need to be modified? For your quick reference here's the stackup on our Gerber files:
[image]

Moreover, I'm looking at some design guidelines made by NCAB ([link](https://www.ncabgroup.com/pcb-design-guidelines/)), and I wonder about to what extent our current Gerber files would comply with these guidelines, say if the Kasli PCB were to be manufactured by NCAB. Thanks!

gkasprow, 2020-11-11:
Just follow the Gerbers. if you modify the stackup, run signal integrity and power integrity simulations to confirm that it  still works.  I cannot say without simulations that it would be fine. I won't take responsibility for any isues that results from modifications.  Simulations take time and I see no reason to do them again. Wahy you cannot simply order boards from existing producers?

closed 2020-11-11 by gkasprow

sbourdeauducq, 2020-11-11:
> Why you cannot simply order boards from existing producers?

Long lead times, and unpredictable coronavirus lockdowns.

So the information in the Gerber is correct? How was this stack-up decided, is it a standard/common NCAB one?

dtcallcock, 2020-11-11:
It sounds like you are suggesting that a large fraction of users should switch hardware suppliers and you want to facilitate that?   If so, I think that such a serious reconfiguration of how the Sinara collaboration works merits some kind of discussion amongst the many stakeholders.

dtcallcock, 2020-11-11:
...like, not here on an obscure closed thread.

dhslichter, 2020-11-11:
Changing a stackup for a board like Kasli is not something that should be undertaken lightly, or without a good purpose.  There is also no such thing as a "standard" stackup, lots of different vendors/groups use different default stackups for different reasons -- and sometimes they change these without warning.  For high performance boards like this, it's not the same as a little hobbyist 2-layer/4-layer for low frequency signals and small currents where these things don't matter as much.  All of the thicknesses (copper, FR-4) listed above are available as standard prepreg or core thicknesses and foil thicknesses, any board house should be able to make them if you specify what you want.

hartytp, 2020-11-11:
I don't want to get into this conversation in any way, but there is perhaps one thing I would say (without really knowing any context/anyone's plans): we've put a lot of time into solving assorted manufacturing issues over the past few years and I would expect that switching to any new supplier will introduce some fresh issues. If I were a user purchasing hardware from a supplier that had previously sold me TS/CTI boards but was now switching suppliers I would want to be explicitly warned about that  unless PI/SI had been run and significant post-production testing carried out. I'd be pissed if I hit unexpected subtle manufacturing issues that wasted my time.

gkasprow, 2020-11-11:
We spent endless hours at the WUT both with CTI and TS to make sure we understand each other and know their PCB , manufacturing process and component vendors. For example, boards are designed with automated spot THT soldering in mind. The mechanic designs take into account manufacturing capabilities - they were trimmed to use UV printer and specific laser/CNC machine. We modified some footprints to improve the production yield, but this was optimized for particular production line, soldering paste and stencil printer. If another HW wendor wants to produce high quality modules, all the steps need to be taken into account.
The part of the OH approach success is at least two stable vendors, so the client has a choice and both vendors compete with price and quality.  This is what we tested already with White Rabbit technology which is remarkable success. I can't really believe that both TS and CTI have long lead times for all modules. The prices are very competetive and both Quartiq and Mlabs get massive discounts.

gkasprow, 2020-11-11:
In case of White Rabbit switch , the project was open source, but it took CTI nearly 2 years to fully implement production, calibration, tests and QC and make the clients happy.  The HW documentation (PCB, mechanics, test procedures) had to be modified to make it compatible with available resources and processes.

gkasprow, 2020-11-11:
One more thing - generic components. Both CTI and TS use the same generic components. That's why we reuse the same parts in the designs so heavily and I'm optimizing every board to limit the BOM. If other generic RC parts are used, for example from unknown vendors, issues may appear. Generic PN does not take into account several factors like C=f(V),  environmental resistance, long-term stability, and other things that may emerge on a long time scale. I have experience with cheap components from occasional vendors. They are good for a product that has 2 years of lifetime. For this reason, we ended cooperation with one international vendor who changed the part vendor without notification, preserving its own PN. That's why we use in all designs only parts from serious vendors. They are not cheap, but they can guarantee quality and long term availability. We are also using parts from trusted vendors only. It's estimated that 30% of the electronic components on the market are counterfeit. It's very tempting to produce things in China, which I also do and use JLPCB service where for a batch of ten, 4 layer boards with assembled components I paid 80$, but I never know which components they use. I do it only for hobby projects.

sbourdeauducq, 2020-11-12:
> Changing a stackup for a board like Kasli is not something that should be undertaken lightly, or without a really good purpose.

I'm not interested in changing the stackup.

> There is also no such thing as a "standard" stackup, lots of different vendors/groups use different default stackups for different reasons

Indeed, so let's change my question "How was this stack-up decided, is it a standard/common NCAB one?" to "is it a default NCAB one?".

> It sounds like you are suggesting that a large fraction of users should switch hardware suppliers and you want to facilitate that? If so, I think that such a serious reconfiguration of how the Sinara collaboration works merits some kind of discussion amongst the many stakeholders.

I want to double-check the correctness of information about the design, namely the published board stackup. Errors in design files happen (https://github.com/sinara-hw/DIO_SMA/issues/8 for example). This isn't "reconfiguring" anything; the hardware was going to be open from the beginning, and there is little value in open but incorrect information that is not used in practice.

> One more thing - generic components. [...]

Thanks for the reminder. I'm not planning to produce anything at a cheap obscure board house in China.

jordens, 2020-11-12:
>  "is it a default NCAB one?"

Can't you answer that yourself? If you need others to help you with preparing production, then you may be misjudging what's required to satisfy your urge to do it yourself. Or you may want to contract and fund them to do it. Conversely, if you can improve and document that production process you are trying to implement, it would certainly be appreciated.

sbourdeauducq, 2020-11-12:
> Can't you answer that yourself?

I'm trying to understand what was done here and how decisions were made. And NCAB does not publish default stackups, so no, I cannot answer that easily. It's a simple question for people who produced with NCAB but not for others.

gkasprow, 2020-11-12:
Normally what you do is you reach the company and ask for materials they have. Different companies use different material vendors. Then you compose your board with these materials.
I usually use Brandner [stack up calculator](https://www.brandner.ee/capabilities/make-your-build-up/), other companies often have similar ones. 
In the case of complex boards, I ask companies like NCAB for their default stack up that satisfies my needs (layers, possible impedance ranges, thickness).
And yes, NCAB provides (on request naturally) default stackups with impedance profiles.
Since TS offers also PCBs, I usually ask them for this information. You can do the same.

sbourdeauducq, 2020-11-12:
@gkasprow thanks.