sinara-hw/Kasli #87: Kasli 2.0 12V input short circuit

sbourdeauducq, 2020-12-31:
I have two Kasli 2.0 that died in exactly the same way: after a few weeks, a short circuit developed on the 12V input. No burned components or other visible damage. Both failures occurred while boards were inside metallic chassis (ESD is very unlikely).

I'm opening this issue here to collect information about this problem.

kaolpr, 2021-01-05:
Have you tried disconnecting the fan? As problems with these have already been reported (#76) maybe it's generating not only vibrations...

sbourdeauducq, 2021-01-05:
This is on Technosystem boards. They came without fan, and we fitted them with another fan model that we used many times before on Kasli 1 without such an issue.
Also, this is a hard short that can be measured with a DMM, without power applied to the board. The MOSFET that controls the fan cannot have voltage above threshold on its gate.

gkasprow, 2021-01-07:
This could be related to the input protection diode.
If you use a power supply with an output voltage that exceeds 12V significantly, you may heat the D1. It may burn after some time and get shorted.
Just remove D1 and see what happenes.

sbourdeauducq, 2021-01-08:
The power supply is the same AKM65US12 that we have been using before on many Kasli 1.1 without problems, and every time I measured their output it was well below the 13.3V breakdown voltage of the SMBJ12A used on all Kaslis. So this scenario does not sound likely.

sbourdeauducq, 2021-01-17:
One of the boards was actually from Creotech.

> Just remove D1 and see what happenes.

Still the same.

sbourdeauducq, 2021-01-17:
> Have you tried disconnecting the fan?

For good measure, I also tried that - still the same, as expected.

maciejprzybysz, 2021-01-18:
We have just checked Kasli v2.0 sent to Technosystem. We found that the cause of the short circuit is C3 capacitor on the TLV62150 DCDC converter input. In the project 10uF / 6.3V capacitor was used. This converter is powered by 12V, so the capacitor should be rated at least 16V. This problem affects all of the Kasli boards manufactured in accordance with [v2.0rc2](https://github.com/sinara-hw/Kasli/releases/tag/v2.0rc2) and [v2.0.1](https://github.com/sinara-hw/Kasli/releases/tag/v2.0.1) release documentation.

[image]

We removed the capacitor from the board. It looks like the short circuit is gone and the board is working fine. Unfortunately, there are no suitable 10uF capacitors in 0402 case on the market, so a design change is required.

dtcallcock, 2021-01-18:
Would stacking 2x 4.7uF on top of each other be a reasonable way of patching existing boards?

sbourdeauducq, 2021-01-19:
> Unfortunately, there are no suitable 10uF capacitors in 0402 case on the market

What makes a capacitor suitable here besides the obvious - X5R dielectric?

> Would stacking 2x 4.7uF on top of each other be a reasonable way of patching existing boards?

Or maybe try to fit a 0603 if the footprint happens to be large enough...

sbourdeauducq, 2021-01-19:
Patch with 0805 capacitor (Murata GRM21BR61C106KE15K) and solder bridges
[image]

Board works again!

akaminska, 2021-01-25:
Thank you all very much for your help with understanding this issue! Attached is a description of a recommended patch
[Kasli_v2_patch_210122.pdf](https://github.com/sinara-hw/Kasli/files/5866415/Kasli_v2_patch_210122.pdf)

commit 8f85d346 (2022-08-05): Small fixes and alternative components - schematics only

- Added SMA washers (closes #74)
- Added note with max and recommended sma input level (closes #83, closes #100)
- Changed C3 to a higher voltage rating (closes #87)
- Fixed #93
- Changed LD14 (closes #67)
- Fixed #96
- Added PCA9539 as an alternative GPIO expander
- Added switching frequency selection to IC7
- Added LT1938 as an alternative to TLV62150
- Added additional I2C bus dedicated to 24AA02\* EEPROM on FPGA pins P19 and P20 (closes #97)
- Removed unused DDR3 DQ termination resistors from FPGA_SDRAM (they weren't placed on the P [... cut]
closed 2022-08-05 by marmeladapk