sinara-hw/Kasli #88: +5v fan output

pkozakiewicz, 2021-02-03:
The fan included in BOM seems to be causing some problems. Some of them have stopped working so far, some of the others are generating vibrations. The mounting clip that could not be mounted easily without special tools is also an issue. What is more, I think that we should not rely on avaliability of those accessories because some day it might cause unnecessary delay. It would be much better if we find some alternative.

My suggestion is to go back to the old heat sink and mount some widely available fan onto it, for example SUNON MF20100V1-1000U-A99. The problem is that 99 percent of fans smaller than 23x23 mm are rated for 5 volts and we have only 12v fan output on the board.

jordens, 2021-02-03:
The sunon fans are good. I'm all for this.

jordens, 2021-02-03:
And Sunon has 12v fans of that size as well which we've been using for years on Kasli v1.1. But the fan needs to be mounted properly. That was the entire purpose of the exercise.

pkozakiewicz, 2021-02-03:
@jordens Could you please give me an example of such a fan? I cannot find anything interesting on Digikey. Wouldn't it be more safe to add +5v output instead?

jordens, 2021-02-03:
https://www.mouser.de/ProductDetail/Sunon/MC25101V2-000U-A99

pkozakiewicz, 2021-02-03:
The one that you proposed is EoL, but I found this 25x25mm : MF25101V3-1000U-A99. It will be slightly larger than sink, but we have no other option. I'll order some and we'll check how it works. 

I'm wondering if gluing it together will look fine for end customers. Maybe it would be better to design our own heatsink.

jordens, 2021-02-03:
That's version 3 of the same fan. We've been gluing them for years. Works very reliably (if done right) and looks fine. Alternatrively now with the space around the FPGA, we could look for an oversized heatsink that we can screw this fan onto.

marmeladapk, 2021-02-03:
We found [this fan](https://www.digikey.pl/products/pl?keywords=MS2006H12D-RSR) but we have a sample size of 1 working continuously at @sbourdeauducq's lab.

sbourdeauducq, 2021-02-03:
Gluing isn't the perfect solution; in particular, it makes the fan difficult to replace. I also wonder if all the weight that the fan puts on the FPGA is ok, we've had a few Kasli boards with cracked FPGA solders (which appeared after shipping) and this could have something to do with it. A large heatsink attached to the PCB sounds more solid and more serviceable.

jordens, 2021-02-03:
FPGA solder joints breaking due to the fan is implausible. The fan is less than 7 g, at 10 G, that's a meager 0.7 N assuming no lever. Even a single solder ball is typically spec'd to hold that and they are ductile. Other things will fail before the fan breaks a proper solder. Not to mention that the current clip fans attach to the FPGA, not the board. That damage you see likely has different origin (several come to mind).

kaolpr, 2021-02-10:
I agree with @sbourdeauducq on gluing fan. Replacement scenario must be considered. On the other hand, replacing fan along with glued radiator can be easier and, when followed by board testing, should be relatively low-risk.

However, it sounds better to have it with clip or attached to PCB. From what I see, finding an integrated radiator and fan having the right size and voltage is somewhat cumbersome, so maybe let's explore PCB attachment. The question is if we can fit mounting holes in the FPGA region? @marmeladapk what do you think?

marmeladapk, 2021-02-11:
I considered it with 2.0 but I couldn't find enough space for it (see #44 and #7). One of the smaller heatsinks with fan that I found was [this](https://cofan-usa.com/pdf/30-2304_REV01.pdf) but it requires enormous holes:

[image]

This and other small heatsinks collide with EEM connectors (we have 6.3mm from south edge of FPGA to north edge of EEM conn). However moving FPGA north is way too much work for the benefit.
[image]

I looked through my notes and it seems that I didn't find a heatsink with fan with that mounting mechanism smaller than 30mm x 30mm. Smaller ones come with a clip. 

Then when it comes to mounting holes, the could be located only on lower left and upper right, although even then I don't think I could preserve proper clearance between traces particularly with clock signals. It's only a 6 layer board after all.

kaolpr, 2021-02-11:
So we're left with two options:

1. glue fan to the radiator and consider detaching radiator-fan set from FPGA a valid replacement method,
2. use radfan with clip - @pkozakiewicz did you find a suitable one for 5V?

pkozakiewicz, 2021-02-11:
@kaolpr I'm somewhere between those two options. Using thermoconductive tape to glue the radiator as we used to do in v1.1 is fine for me. I don't think that removing it generates that much stress to damage the solders. This operation is not performed so often (it happens mostly when replacing FPGA itself). 

However, in the terms of production process, gluing fan to the heatsink seems to be worse option than screwing it. I suggest that we should try to design a new heatsink with threaded holes dedicated to the most popular 20x20 and 25x25 heatsinks. Manufacturing it won't be expensive.

I thought merely of using 5V fan to avoid shipping delays in the future, because tiny fans are much more widely available for +5V than +12V. I will make some stock of 12v fans, but we have to take into account the possible market situation in the future.

kaolpr, 2021-02-11:
I've talked to @jgajewski and we've came up with the solution where thermal tape is used to glue 25x25mm radiator (slightly bigger then the current one) to the FPGA and fan is screwed to the radiator, much alike in the radfan used in the project now.

Changing to 25x25 radiator will make choosing fan [much easier](https://eu.mouser.com/Thermal-Management/Fans-Blowers/_/N-axg88Z1yzvvqx?P=1yzxikxZ1z0jliwZ1z0xjmuZ1z0wae7Z1z0sdx2Z1y94xd8) and we won't have to switch to 5V.

marmeladapk, 2021-02-11:
We used thermal tape in v1.1 and it we decided to move away from this solution: #7. I seem to recall that it was also due to risk (or real incidents, can't remember now) of heatsink falling off in transport.

kaolpr, 2021-02-11:
Ok, so we're left with snap-on option. What would you say on a simple, 3D printed adapter glued onto the radiator (e.g. [this one](https://eu.mouser.com/datasheet/2/596/TS-53230K-C1-R0-1480355.pdf)?

Draft of the adapter is given in the figure. Top holes would contain metal threaded inserts, so screwing and unscrewing fan will not damage the adapter. From bottom two tabs would go in between radiator fins and will be glued to it.
[image]
This solution would allow as using generic 25x25 fan that can be easily replaced and not gluing anything to the FPGA.

kaolpr, 2021-02-19:
Are there any comments on suggested solution?

marmeladapk, 2021-02-22:
@kaolpr I think it's a fine backup plan, but I'd prefer first to get more data on the fan I linked earlier. Hopefully we won't need this at all.