sinara-hw/Kasli #95: Grounding modifications

kaolpr, 2021-12-16:
Depends on the final conclusions of https://github.com/sinara-hw/meta/issues/64.

There are several points to consider:

1. Establish front panel circuit and connect mounting holes to it.
1. Power connector - it was pointed out in #91 that power connector in a current setup connects to the panel and, as a consequence can connect to GND via microUSB port or SFP cage effectively diminishing effect of EMI filters (FL1, FL2). As a solution we could pull the connector back and increase FP cutout, so that insulating sleeve can fit between connector and front panel. This would mean lack of nut, but should not be a problem.
2. microUSB shield should be connected to FP circuit and disconnected from GND.
3. SMA stays as it is.
4. Connect SFP cages to FP circuit.

gkasprow, 2021-12-17:
We should do it with properly characterized power sources. Both stand-alone and 3U ones.

kaolpr, 2021-12-17:
What does power source characterization has to do with it? Isn't it only the matter of consequent separation of chassis circuit from the module's ground?

gkasprow, 2021-12-17:
Not only, but It also creates the common mode noise that is propagated across the crate via a ground loop it creates with the AC network.
There are many more such loops that are not obvious.