sinara-hw/Kasli #100: External clock reference power limit

Spaqin, 2022-07-26:


marmeladapk, 2022-07-26:
It was not updated for v2. I'll update the wiki and add a note to
schematics (#83).

sbourdeauducq, 2022-07-26:
This means using Clocker as clock source requires an attenuator.

hartytp, 2022-07-26:
> 10dBm would mean 2Vp-p w/ 50ohm impedance (and that is doubled by the transformer at the SMA input).

Checking we're on the same page, but the 2:1 transformer is a sqrt(2) step up on the voltage, not a factor of 2. Or, is that what you meant?

hartytp, 2022-07-26:
> We found this discrepancy while trying to connect Clocker output to Kasli clock input. Since Clocker output is 800mVp-p with 50ohm impedance, we thought it could cause damage to the FPGA.

nit pick but for the sake of accuracy...agreed that over-driving an input isn't a good idea, but it's worth noting that (unless I'm mistaken) the numbers you quote there are the recommended operating conditions, not the abs max (no damage) ratings. Usually the damage threshold is quite a bit higher than these numbers.

hartytp, 2022-07-26:
> This means using Clocker as clock source requires an attenuator.

Throwing a couple of options out there if only to discount them:
(a) on the next hw revision we put a clock buffer in front of the FPGA. This adds a bit of additional robustness at the cost of a small amount of additional power dissipation. Am I right in thinking that if we used a 1:2 fanout buffer with the second output connected to CLKIN2 on the 5324 we no longer need to route this signal via the FPGA. May not actually be worth the effort thought...
(b) make a version of clocker which uses a LVDS buffer. Since nothing in Sinara actually uses LVPECL the current design is perhaps not ideally matched. The one caveat here is that the extra uumph is often useful when driving long cable runs which add additional attenuation.

gkasprow, 2022-07-26:
we may also add a Schottky diode limiter

commit 8f85d346 (2022-08-05): Small fixes and alternative components - schematics only

- Added SMA washers (closes #74)
- Added note with max and recommended sma input level (closes #83, closes #100)
- Changed C3 to a higher voltage rating (closes #87)
- Fixed #93
- Changed LD14 (closes #67)
- Fixed #96
- Added PCA9539 as an alternative GPIO expander
- Added switching frequency selection to IC7
- Added LT1938 as an alternative to TLV62150
- Added additional I2C bus dedicated to 24AA02\* EEPROM on FPGA pins P19 and P20 (closes #97)
- Removed unused DDR3 DQ termination resistors from FPGA_SDRAM (they weren't placed on the P [... cut]
closed 2022-08-05 by marmeladapk