sinara-hw/Kasli #101: Recommendations to the new revision regarding components supply.

pkozakiewicz, 2022-08-01:
@marmeladapk Below you can find a list of current shortages:

Si5324C-C-GM
24AA025E48-I/SN
TCA9548ARGER
MCP23017-E/SS
549BACB001937ABG

Please consider finding alternatives as you redesign Kasli.

sbourdeauducq, 2022-08-01:
> Si5324C-C-GM
> 549BACB001937ABG

Most parts from Silabs/Skyworks seem to be unobtainium these days. Do you know what is going on with that company?
Replacing those looks like it's going to be a pain.

> TCA9548ARGER

PCA9547 is a viable and available alternative. It is electrically and mechanically compatible and the board does not need to be modified. We use it on some Kasli-SoC. It would simply need porting support from zynq-rs.

pkozakiewicz, 2022-08-01:
> Do you know what is going on with that company?

There's been an acquisition of one company by another and there have been supply chain issues since then. Many chips are claimed to be available only on January 2024. Some could be sourced from the brokers but the prices are approx 20-50x higher.

marmeladapk, 2022-08-01:
FOR TCA9548 I'm going to suggest MAX7356ETG as an alternative, it's somewhat available now. However adding support for PCA9547 would be nice.

marmeladapk, 2022-08-01:
@sbourdeauducq I'm going to add PCA9539 as alternative to MCP23017-E/SS (I'll try to fit both footprints), could you add support for that? MCP23017 price also skyrocketed and PCA9539 seems to be available at the moment and there are pin compatible alternatives.

sbourdeauducq, 2022-08-01:
In principle yes but we should check the compatibility carefully and also how to autodetect which chip is present on the board. It would be annoying to require different firmware builds. For the PCA9547 we can do it by reading the register values after reset.

marmeladapk, 2022-08-01:
PCA9539 uses different address so you can check which one is present on the board.

pkozakiewicz, 2022-08-01:
PS It would also be good to have an alternative for S25FL128SAGMFIR01.

marmeladapk, 2022-08-05:
24AA025E48-I/SN, TCA9548ARGER, MCP23017-E/SS now have alternatives, clock chips unfortunately no.

closed 2022-08-05 by marmeladapk

pkozakiewicz, 2022-08-05:
And what about flash memory?

marmeladapk, 2022-08-05:
There are still some memory configurations that are available. Unfortunately I can't use another package as only SOIC package has VIO pin (unless we move to BGA packages). I also can't fit another footprint in there.