sinara-hw/Kasli #102: Replacement fan info

TopQuark12, 2022-09-16:
We had kasli cooling fans fail on some of our boards, and the original fan was acoustically a bit obnoxious. 

We looked for a replacement, and [CFM-2510B-1100-218](https://www.mouser.hk/ProductDetail/CUI-Devices/CFM-2510B-1100-218?qs=GedFDFLaBXHxyJdzwff0Gg%3D%3D) from CUI Devices works as a replacement for kasli and kasli soc cooling fan. 

The fit of this fan on kasli is a bit tight, care must be taken to make sure the screws do screw into the heatsink fins, but it does work. Ideally we would use 20x20 mm fans, but choices are limited with that form factor, especially from good manufacturers. The fitment of this fan is better on kasli-soc with its larger heatsink. M2x16 screws are needed for this replacement fan, and the fan comes un-terminated, so a crimper and crimps are needed too. We only had JST XH connectors on hand so that's what we used, but ideally the matching molex connectors should be used.

marmeladapk, 2022-10-12:
For future reference: https://github.com/sinara-hw/Kasli/issues/88#issuecomment-772604088

lriesebos, 2024-03-13:
I can also confirm that we have seen multiple of these radian fans break. having a better fan to start with would be great. the recommended replacement fan works fine, but it is not really a great fit on the existing heatsink too.