sinara-hw/Kasli #109: Add P12V0 switch for EEMs

marmeladapk, 2024-04-26:
[image]

Somewhere here, and switch P12V0 for EEMs only after FPGA is programmed.

gkasprow, 2024-04-26:
That wont work because many boards have independent supply. But in DIOT we might use servmod line to control supplies

gkasprow, 2024-04-27:
Most boards don't use external power connectors, but we have a few that do:
HVAMP32
HVAMP8
SILPA
Thermostat EEM
Stabilizer
Magneto (new one)
HV_PSU
VHF_AMP(new one)

Having even one of them in the crate will make such switching impossible and cause an unintended flow of current.

gkasprow, 2024-04-27:
> 
> 
>     HVAMP32
>     HVAMP8
> 
> AFAIK those just have unidirectional inputs and cannot cause the original latchup problem?

But they short 12V from EEM to external 12V and will supply other boards anyway

> 
>     SILPA
> 
> Those are not usually connected to Kasli and run independently.

It can also work as EEM; we even have an ARTIQ driver.

marmeladapk, 2024-05-06:
Still these cards will be in minority of setups, so we could protect most of them. Later in DIOT we won't need that anyway.

PR #114 (2025-09-04): Kasli v2.1.0rc1
# Description
This PR fixes issues mentioned in #113 v2.1.0 review.

## List of Changes
### Schematics
1. Add nettie symbols to EEM_FLT_N and P3V3 Net
2. Update the schematics rev number

### PCB Layout
1. Update the front panel side board outline according to https://github.com/sinara-hw/DIOT_PSU_integrated/issues/3
2. Update the copper thieving pattern as in [DIOT Template](https://github.com/sinara-hw/DIOT-Template)
3. Reduce the size of P12V0_EEM polygon in SignalLayer1
4. Set P3V3 polygon on bottom layer NOT to contain dead copper
5. Add thermal reliefs or solid copper reg [... cut]